nickgammon/arduino_sketches #7: Missing F() makro in board programmer

NicoHood, 2015-01-25:
https://github.com/nickgammon/arduino_sketches/blob/master/Atmega_Board_Programmer/Atmega_Board_Programmer.ino#L480
https://github.com/nickgammon/arduino_sketches/blob/master/Atmega_Board_Programmer/Atmega_Board_Programmer.ino#L595
https://github.com/nickgammon/arduino_sketches/blob/master/Atmega_Board_Programmer/Atmega_Board_Programmer.ino#L681

I am also wondering if we should put this into a special function to make it more clear (and easier to modify) Maybe something with a better API to also add BootLoaders for Micro and Pro Micro for example.
https://github.com/nickgammon/arduino_sketches/blob/master/Atmega_Board_Programmer/Atmega_Board_Programmer.ino#L519-L542

And btw a little note here:
My modified version uses 99% of the Unos memory. One guy got a verification error at byte 0x7800. Seems that he has no Optiboot installed. For further development you should keep in mind to no exceed this size for an uno. Also the sketch does not compile by default with a Leonardo. But you have the settings, which are nice. Should be fine, just wanted to mention it. Maybe you could default outcomment some bootloader but add some more for specific MCUs like Pro Micro, Micro etc.

NicoHood, 2015-01-25:
On top of that I suggest if writing the bootloader failed, that we dont change the fuses then and abort:
https://github.com/nickgammon/arduino_sketches/blob/master/Atmega_Board_Programmer/Atmega_Board_Programmer.ino#L633-L644

NicoHood, 2015-01-25:
Another bug I found:
Try to upload the sketch, dont connect any wires and try to start the sketch. It will fail. Enter C and try again. It will hang because of the first SPI transfer in the startprogramming loop. And it would also be better to set the variable "confirm" to zero in this function to ensure its not accidentally 0x53.

nickgammon, 2015-01-25:
> On top of that I suggest if writing the bootloader failed, that we dont change the fuses then and abort:

How do you copy a batch of lines into a comment?

Anyway:

```
  if (errors == 0)
    Serial.println (F("No errors found."));
  else
    {
    Serial.print (errors, DEC);
    Serial.println (F(" verification error(s)."));
    if (errors > 100)
      Serial.println (F("First 100 shown."));
    return;  // don't change fuses if errors
    }  // end if
```

Note the "return" statement. The fuses are not changed if there are errors.

nickgammon, 2015-01-25:
> It will hang because of the first SPI transfer in the startprogramming loop. 

I can't reproduce that.

> And it would also be better to set the variable "confirm" to zero in this function to ensure its not accidentally 0x53.

What do you mean? It is a local variable and is always written to (by SPI.transfer) before being tested.

nickgammon, 2015-01-25:
I made your suggested changes to the F macro.

nickgammon, 2015-01-25:
> One guy got a verification error at byte 0x7800. Seems that he has no Optiboot installed. 

I moved the configuration lines higher up, so it is more obvious what to change if the sketch is too big.

NicoHood, 2015-01-27:
> How do you copy a batch of lines into a comment?
hm? click the first line, then shift and then click the second line?

> I can't reproduce that.
I think I had a 20-40cm wire (6 flatten wires together with that 6 pin header) attached. Maybe that made the pin floating?

NicoHood, 2015-01-27:
Ahm just a simple question:

Cant we put the array into progmem?
const PROGMEM signatureType signatures [] =

Haven't tested this, and can't right now. But this would save a lot of ram. If it doesn't work, do you know why?

nickgammon, 2015-01-27:
> hm? click the first line, then shift and then click the second line?

I mean like this: You have a link to the lines, you didn't copy the lines:

https://github.com/nickgammon/arduino_sketches/blob/master/Atmega_Board_Programmer/Atmega_Board_Programmer.ino#L633-L644

> I think I had a 20-40cm wire (6 flatten wires together with that 6 pin header) attached. Maybe that made the pin floating?

I don't see how a floating pin would make the sketch hang.

nickgammon, 2015-01-27:
> Haven't tested this, and can't right now. But this would save a lot of ram. If it doesn't work, do you know why?

RAM isn't particularly short is it in this sketch?

NicoHood, 2015-01-27:
> I mean like this: You have a link to the lines, you didn't copy the lines:

ahm i didnt want to post any code here. seems you already got it correct and i missed the return.

> I don't see how a floating pin would make the sketch hang.

me neither. have to check it again. compiling with 1.6 rc2 btw

> RAM isn't particularly short is it in this sketch?

no, just asking. 1333 bytes or something like that i remember is quite a lot. Just wondering if we could improve this with some simple modifications or if this would make it complicated. I am currently thinking of some ideas to save PCINT function pointers and this inspired me.

nickgammon, 2015-01-27:
It's simple enough. I put the table into PROGMEM in a variation of the sketch (I think the one that reports board types). But if there are 600 bytes of RAM free it just seems to add complexity for no gain, unless you are planning to run it on a board with a lower spec.

> me neither. have to check it again. compiling with 1.6 rc2 btw

I'm wondering if your (own) version is hanging, rather than my official one. Can you give a link to yours? For one thing your question about the "confirm" variable doesn't seem to apply to my version.

NicoHood, 2015-02-18:
Sorry for the long delay of my response.

I just tested it again. I used an Arduino Uno Clone board and uploaded your sketch (atmega board programmer). No wires attached to the SPI pins (nor pin 10-13).

Its loads and fails. Then it asks again. Then i type c and it stucks.
```
`Attempting to enter programming mode ...
Atmega chip programmer.
Written by Nick Gammon.
Version 1.30
Compiled on Feb 18 2015 at 11:03:37
Attempting to enter programming mode ......................................................
Failed to enter programming mode. Double-check wiring!
Programming mode off.
Type 'C' when ready to continue with another chip ...
Attempting to enter programming mode ...
```

Used IDE 1.6 Win7/8.1 to compile. (maybe you should add the IDE version to the beginning of the sketch as well).

nickgammon, 2015-02-18:
It worked for me. Are you using my exact sketch or a variation of it?

NicoHood, 2015-02-19:
Just downloaded your version yesterday. No modification. Using the newest IDE 1.6 (not 1.0.6)!

nickgammon, 2015-02-21:
Fixed here: https://github.com/nickgammon/arduino_sketches/commit/4b0730bef96b8ebab0419a7cb88ee826b55948b2

closed 2015-02-21 by nickgammon

NicoHood, 2015-02-24:
Working as expected now :)