nickgammon/arduino_sketches #16: Extranous gpl.txt and md5.{c,h} in Atmega_Board_Programmer

matthijskooijman, 2015-04-09:
I was just looking at the Atmega_Board_Programmer sketch. It has an md5 implementation included, but unless I'm very much mistaken, it is not used at all?

Furthermore, there is a gpl.txt in there, but the sketch seems to be licensed under a more liberal license. Does the gpl perhaps apply to the bootloaders? If so, some clarification is needed.

matthijskooijman, 2015-04-09:
Ah, it s eems md5.c is licensed under the gpl, so that's probably what the gpl.txt is for. I have a commit to remove both of them here, just let me know if you want them removed, then I'll send a PR.

nickgammon, 2015-04-09:
Version 1.11 removed the MD5 stuff to make room for more board types:

> // Version 1.11: Added support for Atmega1280. Removed MD5SUM stuff to make room.

You are right it isn't needed. I'll do it now and push the results. I won't change the version number as it doesn't really affect anything material.

The MD5 stuff is still used in other sketches in that repository.

nickgammon, 2015-04-09:
Done now. https://github.com/nickgammon/arduino_sketches/commit/9854db4fea03be2f74cf95c55cf6aa94561d3508

matthijskooijman, 2015-04-09:
Cool, thanks.

closed 2015-04-09 by matthijskooijman