nickgammon/arduino_sketches #17: Supporting atmega256rfr2 / Pinoccio Scout

matthijskooijman, 2015-04-09:
I've been working on an official update for the bootloader for the Pinoccio Scout board. Previously, we've distributed a custom sketch based on your code for updating. Now, I'm wondering if you would be willing to have direct support for the rfr2 / Pinoccio Scout in your Atmega_Board_Programmer sketch (possibly disabled by default).

If the answer is yes, I'll submit a PR with the relevant code, and supply a bit of documentation and pictures to include here (assuming that's still the most recent documentation location): http://www.gammon.com.au/forum/?id=11635

nickgammon, 2015-04-09:
That seems OK to me. I presume you won't be changing the bootloader frequently? I think we'd disable it by default, the sketch size is getting quite large as it is.

matthijskooijman, 2015-04-17:
I don't expect frequent changes. We're releasing a new version with a bugfix soon, for which I'd like to use your sketch as the primary distribution method for our users. After that, we'll probably do a big overhaul for the v1.1 or v1.2 board, but that should also result in just one update, probably later this year or early next year.

Disabling it by default sounds fine to me. Perhaps it would be good to show a message that states that the bootloader is disabled at compiletime, instead of the generic "No bootloader support for this device.". Perhaps a bit of restructuring can also slightly clean up the #ifdef stuff in the signatures array, that I personally don't find very elegant right now.

commit f8ef6263 (2015-04-21): Allow flashing a 256rfr2 bootloader

This uses the bootloader from the Pinoccio project, version v1.0a:

	https://github.com/Pinoccio/hardware-pinoccio/tree/v1.0a/firmware/bootloader

To save space, this bootloader is disabled by default.

This fixes #17.
closed 2015-04-21 by nickgammon