nickgammon/arduino_sketches #20: Arduino-COMBINED-dfu-usbserial-atmega16u2-Uno-Rev3 question

dmjlambert, 2015-07-04:
Hi,

I have more of questions than an issues, if you don't mind.   I noticed in the Atmega_Board_Programmer file bootloader_atmega16u2.h that the hex code which has been put in the array may have come from the file Arduino-COMBINED-dfu-usbserial-atmega16u2-Uno-Rev3.hex and there is a comment saying Loader start: 3000, length: 4096.   So I suppose just the DFU bootloader extracted from Arduino-COMBINED-dfu-usbserial-atmega16u2-Uno-Rev3.hex to form the bootloader_atmega16u2.h file, and the USB-serial program was skipped over when the .h file was created, is that correct?

I am wondering if there is potential for including the whole Arduino-COMBINED-dfu-usbserial-atmega16u2-Uno-Rev3.hex in the .h file so when programming the DFU bootloader the USB-serial program can also be included.  It seems that would be harmless since the chip is getting erased anyway, and it could be useful because it would give a one-step "factory reset" of the UNO.  

I understand there is a memory gap between the USB-serial program and the bootloader which would make a combined image a big hog of the programmer's flash unless there was some sort of gap-filler or compression logic in the programmer.

Thanks,
David

nickgammon, 2015-07-04:
The basic problem was what you surmised. To make the sketch upload the "low part" and also the bootloader would have been quite a bit more complexity - plus the size of the code itself. It didn't seem particularly worth it. There is a way of getting the main code (other than the bootloader) onto the chip, and in any case I doubt people need to reflash the Atmega16U2 that often.

NicoHood, 2015-07-04:
Thats what I also suggested. It would be a good option to also burn the whole program with usb-Serial.

closed 2018-04-03 by nickgammon