nickgammon/arduino_sketches #21: MOSFET Vgs(max)

robert-rozee, 2015-08-08:
hi,
    your webpage at http://www.gammon.com.au/motors was recently referenced in a thread i've been following, and i notice a design oversight that you seem to be frequently repeating.

when using a P-channel MOSFET as a high-side switch, you often drive the gate with an NPN transistor but fail to include a suitable zener diode (6v8 or similar) to protect the MOSFET from excessive gate voltage. generally Vgs should not exceed the range +/- 8v. if just using an NPN transistor to drive directly into the gate, without suitable gate protection, it is important that the supply voltage connected to the MOSFET's source should not exceed 8v or so.

(the below circuit was for use with a MOSFET that had a slightly higher Vgs(max)
[image]

cheers,
rob   :-)

nickgammon, 2015-08-08:
The one I referenced on my page (FQP47P06) has a Vgs(max) of +/- 25V, so that particular circuit should be OK, shouldn't it? However what you are saying is that, in principle, a zener diode along the lines of 6V8 would be a sound idea?

nickgammon, 2015-08-08:
Doesn't a zener require some current to activate? Your circuit would only have about 2 mA through it if V(load) was, say, 24 V. 

What about the suggestion half-way down on http://electronics.stackexchange.com/questions/169211/using-npn-bjt-with-a-p-channel-mosfet-to-switch-3-3v-to-24v ?

robert-rozee, 2015-08-09:
if Vgs(max) is sufficiently high for the part being used, then no zener is required. but that would require checking the datasheet of the specific part, something few folks seem to do. i've been told that about 8v is a safe assumption - for the DMP2305U it is +/- 8 volts. people also seem to draw little distinction between MOSFETs of significantly differing sizes, hence erring on the safe side is often the best practice, with perhaps a note added after the schematic saying under what conditions the zener may be left out.

you are right about zeners requiring at least a certain current before they start regulating - but with too low a current through a zener, it will tend to regulate to a LOWER voltage than rated, hence still protecting the gate of the MOSFET. in general, i've found a few milliamps is sufficient, though there seems to be no universal answer to what current is recommended.

placing a zener in series with the gate will, in theory work, but you will need to ensure (a) sufficient current flows in the 'off' state, and (b) that the zener voltage is 'just right' relative to the load supply to ensure that there is still sufficient gate voltage when you want the MOSFET to be 'on'. the second schematic on the stackexchange page you link to is closer to mine, but has the fault (in my opinion) that R1 and Q1 are in reverse order - R1 should be in the collector, and there should be a base current limiting resistor. the author has tried to do without a base resistor, but has just made the circuit operation far more dependent on supply voltages and component values.

nickgammon, 2015-08-09:
I was wondering about the base resistor, and trying to get my mind around the exact way that would work.

robert-rozee, 2015-08-09:
upon reading more of the text, and delving into a few dusty corners of the mind... it may in fact be a quite clever bit of design!

if V1 = 3.3v, and the Vbe of T1 is 0.6v, then R1 must have a 2.7v drop across it (when V1 is present) and hence be passing 2.7/104 =  25mA. this current may flow in either base or collector, but will tend to balance between the base and collector based upon the the gain of T1. more-or-less, the majority of the current will flow into the collector, while just a fraction will flow into the base.

now, the down side is that the control pin (V1) may see a very brief 25mA draw when first turned on, and the current limit set is dependent on the V1 voltage. but these are not biggies. i would tend to use a much lower zener current myself, making R1 something like 1k.

robert-rozee, 2015-08-09:
the one down side is that if V2 is absent, then the current through R1 will be sourced entirely from V1 - ie, as it stands, if V2 is missing there will be a 25mA current draw from V1 T1's B-E junction will act like a simple diode). for most small micros, this is a tad excessive, especially if several pins are driving similar circuits.

a 3k3 or so base resistor would largely mitigate this problem, without having a material effect on the zener current.

nickgammon, 2015-08-16:
After various discussions on the Arduino forum, I think the secondary schematic on http://www.gammon.com.au/motors should solve this issue.

[image]

robert-rozee, 2015-08-16:
that looks good. the important thing is that Vgs(max) is not exceeded excessively. i had a quick look at the arduino forum discussion (http://forum.arduino.cc/index.php?topic=340751.0) and most of the issues seem to have been well covered.

i guess the biggest problem is people doing a google search and just grabbing (and using) the first  schematic they find, without going into any detail. alas, there is no (simple) single solution that works in all cases; it depends on Vgs(max), the minimum Vgs that will achieve a full turn-on of the MOSFET, the supply for the load (5v, 12v, 24v, etc), and the rate of switching (PWM brightness control, or just turning on something whenever a button is pressed).

nickgammon, 2015-08-16:
Well, I think it is covered now. I have a schematic for 5 V, 12 V, and 24 V, all different. :)

closed 2015-08-16 by nickgammon