nickgammon/arduino_sketches #23: Bootloader programmer fails to enter programming mode with second chip.

amowry, 2015-08-27:
The bootloader programmer is working great for the first chip, but with the second one it fails to enter programming mode until I restart the programmer. Any suggestions for this? Thanks! The second chip is identical to the first. Note: If instead of a second chip I bootload the first chip again after a successful attempt, it works fine.

Attempting to enter programming mode ...
Entered programming mode OK.
Signature = 0x1E 0x95 0x0F 
Processor = ATmega328P
Flash memory size = 32768 bytes.
LFuse = 0x62 
HFuse = 0xD9 
EFuse = 0xFF 
Lock byte = 0xFF 
Clock calibration = 0x8E 
Type 'L' to use Lilypad (8 MHz) loader, or 'U' for Uno (16 MHz) loader ...
Using Uno Optiboot 16 MHz loader.
Bootloader address = 0x7E00
Bootloader length = 512 bytes.
Type 'Q' to quit, 'V' to verify, or 'G' to program the chip with the bootloader ...
Clearing 'Divide clock by 8' fuse bit.
Fixing low fuse setting ...
Attempting to enter programming mode ...
Entered programming mode OK.
Erasing chip ...
Writing bootloader ...
Committing page starting at 0x7E00
Committing page starting at 0x7E80
Committing page starting at 0x7F00
Committing page starting at 0x7F80
Written.
Verifying ...
No errors found.
Writing fuses ...
LFuse = 0xFF 
HFuse = 0xDE 
EFuse = 0xFE 
Lock byte = 0xEF 
Clock calibration = 0x8E 
Done.
Programming mode off.
Type 'C' when ready to continue with another chip ...
Attempting to enter programming mode ...  
((sticks here and never enters programming mode))

nickgammon, 2015-08-27:
Which version do you have? You should see:

```Attempting to enter ICSP programming mode ...```

You should also see a version, and date/time compiled, at startup.

amowry, 2015-08-27:
Thanks Nick. I don’t think I ever see Attempting to enter ICSP programming mode ...

Here’s the complete log from one successful attempt, with the version number etc.

Atmega chip programmer.
Written by Nick Gammon.
Version 1.33
Compiled on Aug 24 2015 at 12:04:10 with Arduino IDE 10602.
Attempting to enter programming mode ...
Entered programming mode OK.
Signature = 0x1E 0x95 0x0F 
Processor = ATmega328P
Flash memory size = 32768 bytes.
LFuse = 0xFF 
HFuse = 0xDE 
EFuse = 0xFE 
Lock byte = 0xEF 
Clock calibration = 0x92 
Type 'L' to use Lilypad (8 MHz) loader, or 'U' for Uno (16 MHz) loader ...
Using Uno Optiboot 16 MHz loader.
Bootloader address = 0x7E00
Bootloader length = 512 bytes.
Type 'Q' to quit, 'V' to verify, or 'G' to program the chip with the bootloader ...
Erasing chip ...
Writing bootloader ...
Committing page starting at 0x7E00
Committing page starting at 0x7E80
Committing page starting at 0x7F00
Committing page starting at 0x7F80
Written.
Verifying ...
No errors found.
Writing fuses ...
LFuse = 0xFF 
HFuse = 0xDE 
EFuse = 0xFE 
Lock byte = 0xEF 
Clock calibration = 0x92 
Done.
Programming mode off.
Type 'C' when ready to continue with another chip ...

From: Nick Gammon 
Sent: Thursday, August 27, 2015 1:59 PM
To: nickgammon/arduino_sketches 
Cc: amowry 
Subject: Re: [arduino_sketches] Bootloader programmer fails to enter programming mode with second chip. (#23)

Which version do you have? You should see:

Attempting to enter ICSP programming mode ...

You should also see a version, and date/time compiled, at startup.

nickgammon, 2015-08-27:
Well, you should. See this page: https://github.com/nickgammon/arduino_sketches/blob/master/Atmega_Board_Programmer/ICSP_Utils.ino

```
// put chip into programming mode    
bool startProgramming ()
  {
    
  Serial.print (F("Attempting to enter ICSP programming mode ..."));
```

Are you using ICSP mode (ie. through the 6-pin header)? Or are you using high-voltage programming?

amowry, 2015-08-27:
For some reason I didn’t have all the files. I downloaded everything again, and it’s working great now. Thanks so much for all your help!

 

From: Nick Gammon 
Sent: Thursday, August 27, 2015 2:45 PM
To: nickgammon/arduino_sketches 
Cc: amowry 
Subject: Re: [arduino_sketches] Bootloader programmer fails to enter programming mode with second chip. (#23)

Well, you should. See this page: https://github.com/nickgammon/arduino_sketches/blob/master/Atmega_Board_Programmer/ICSP_Utils.ino

// put chip into programming mode    
bool startProgramming ()
  {

  Serial.print (F("Attempting to enter ICSP programming mode ..."));
Are you using ICSP mode (ie. through the 6-pin header)? Or are you using high-voltage programming?

nickgammon, 2015-08-27:
Glad it works. Looks like you found a bug which was previously fixed. I seem to recall one along those lines before.

closed 2015-08-27 by nickgammon

kxling, 2017-10-27:
borrowing this post as it is related to an issue i am currently facing

Below shows  what is displayed on serial monitor. what is the possible cause? my 'programming' board is arduino uno and my 'target' board is arduino pro mini 5V. i connected using manual programming cable. i successfully upload the sketch, so i doubt its the connection problem

Atmega chip programmer.
Written by Nick Gammon.
Version 1.37
Compiled on Oct 27 2017 at 17:06:38 with Arduino IDE 10803.
Attempting to enter ICSP programming mode ......................................................
Failed to enter programming mode. Double-check wiring!
Type 'C' when ready to continue with another chip ...