nickgammon/arduino_sketches #24: Compiling Error

Jim4Beam, 2015-08-28:
Atmega Board Detector on Arduino IDE 1.6.5 under Ubuntu 14.04 I get:
Arduino: 1.6.5 (Linux), Platine: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

Verwende die Bibliothek SPI im Ordner: /home/jose/Jo/100_Prog/arduino-1.6.5/hardware/arduino/avr/libraries/SPI 
Verwende die Bibliothek Atmega_Board_Detector im Ordner: /home/jose/Jo/100_Prog/arduino-1.6.5/libraries/Atmega_Board_Detector (legacy)

/home/jose/Jo/100_Prog/arduino-1.6.5/hardware/tools/avr/bin/avr-gcc -c -g -Os -Wall -Wextra -ffunction-sections -fdata-sections -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -I/home/jose/Jo/100_Prog/arduino-1.6.5/hardware/arduino/avr/cores/arduino -I/home/jose/Jo/100_Prog/arduino-1.6.5/hardware/arduino/avr/variants/mega -I/home/jose/Jo/100_Prog/arduino-1.6.5/hardware/arduino/avr/libraries/SPI -I/home/jose/Jo/100_Prog/arduino-1.6.5/libraries/Atmega_Board_Detector /tmp/build7606571625362604800.tmp/md5.c -o /tmp/build7606571625362604800.tmp/md5.c.o 
/tmp/build7606571625362604800.tmp/md5.c: In function 'md5_update':
/tmp/build7606571625362604800.tmp/md5.c:190:9: warning: implicit declaration of function 'memcpy' [-Wimplicit-function-declaration]
         memcpy( (void *) (ctx->buffer + left),
         ^
/tmp/build7606571625362604800.tmp/md5.c:190:9: warning: incompatible implicit declaration of built-in function 'memcpy' [enabled by default]
/tmp/build7606571625362604800.tmp/md5.c:207:9: warning: incompatible implicit declaration of built-in function 'memcpy' [enabled by default]
         memcpy( (void *) (ctx->buffer + left),
         ^
/home/jose/Jo/100_Prog/arduino-1.6.5/hardware/tools/avr/bin/avr-g++ -c -g -Os -Wall -Wextra -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -I/home/jose/Jo/100_Prog/arduino-1.6.5/hardware/arduino/avr/cores/arduino -I/home/jose/Jo/100_Prog/arduino-1.6.5/hardware/arduino/avr/variants/mega -I/home/jose/Jo/100_Prog/arduino-1.6.5/hardware/arduino/avr/libraries/SPI -I/home/jose/Jo/100_Prog/arduino-1.6.5/libraries/Atmega_Board_Detector /tmp/build7606571625362604800.tmp/Atmega_Board_Detector.cpp -o /tmp/build7606571625362604800.tmp/Atmega_Board_Detector.cpp.o 
In file included from Atmega_Board_Detector.ino:91:0:
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
   };  // end of signatures
   ^
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
/tmp/build7606571625362604800.tmp/Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
Atmega_Board_Detector.ino: In function 'void readBootloader()':
Atmega_Board_Detector.ino:252:23: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
Atmega_Board_Detector.ino:286:30: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
In file included from Atmega_Board_Detector.ino:92:0:
/tmp/build7606571625362604800.tmp/General_Stuff.h:37:72: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
 #define NUMITEMS(arg) ((unsigned int) (sizeof (arg) / sizeof (arg [0])))
                                                                        ^
Atmega_Board_Detector.ino:295:25: note: in expansion of macro 'NUMITEMS'
Atmega_Board_Detector.ino: In function 'void readProgram()':
Atmega_Board_Detector.ino:323:23: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
In file included from Atmega_Board_Detector.ino:92:0:
Atmega_Board_Detector.ino: In function 'void getSignature()':
[... cut ...]
/tmp/build7606571625362604800.tmp/Atmega_Board_Detector/md5.c.o: In function `md5_starts':
/home/jose/Jo/100_Prog/arduino-1.6.5/libraries/Atmega_Board_Detector/md5.c:40: multiple definition of `md5_finish'
/tmp/build7606571625362604800.tmp/md5.c.o:/tmp/build7606571625362604800.tmp/md5.c:40: first defined here
collect2: error: ld returned 1 exit status
Multiple libraries were found for "md5.h"
 Used: /home/jose/Jo/100_Prog/arduino-1.6.5/libraries/Atmega_Board_Detector
 Not used: /home/jose/Jo/100_Prog/arduino-1.6.5/libraries/Atmega_Self_Read_Signature
Fehler beim Kompilieren.

nickgammon, 2015-08-28:
Can you turn those messages into a code block please? Use three back-ticks before and afterwards.

\`\`\`
(put messages here)
\`\`\`

nickgammon, 2015-08-28:
It compiles for me without errors using IDE 1.6.5, on Ubuntu 14.04, for the Mega2560.

You could try adding to the start of md5.h file:

```
#include <string.h>
```

Jim4Beam, 2015-08-29:
Thank you. I will try it out tomorrow.

Jim4Beam, 2015-08-30:
It is working. I had included the Library files twice. My fault. 
Thank you very much for your quick response!

closed 2015-08-30 by Jim4Beam