nickgammon/arduino_sketches #25: problem compiling Atmega_Hex_Uploader

mythral0, 2015-09-16:
Hi,
I am using IDE 1.6.5 and I cannot compile Atmega_Hex_Uploader
it complains about unable to find SPI.h
I have managed to use your other apps and have also managed to load a 'boot loader' onto a chip.
I have tried reinstalling the ide, but with no luck.
I have also written a dummy sketch using the afending library and that seems to compile fine.
Please help.

KND RGS

Mo.

nickgammon, 2015-09-16:
It compiled OK for me, but the include for SPI.h was missing. I've tidied up various things, try downloading the new version now.

See: https://github.com/nickgammon/arduino_sketches/commit/4654efe213efbef862d

mythral0, 2015-09-17:
Hi,
Many thanks for a quick response.
Unfortunately, I am now getting a different error, please see below.
I wanted to use Atmega_Hex_Uploader to modify the fuses, however, I have achieved this by modifying your Atmega_Board_Programmer code with the fuse values I wantedd and then uploaded the bootloader to the chip.

Arduino: 1.6.5 (Windows 7), Board: "Arduino/Genuino Uno"

In file included from C:\Users\vallim\Documents\Arduino\libraries\SdFat/SdFat.h:26:0,
                 from Atmega_Hex_Uploader.ino:120:
C:\Users\vallim\Documents\Arduino\libraries\SdFat/SdSpiCard.h:302:8: warning: unused parameter 'spi' [-Wunused-parameter]
   bool begin(m_spi_t* spi, uint8_t chipSelectPin = SS,
        ^
C:\Users\vallim\Documents\Arduino\libraries\SdFat/SdSpiCard.h:302:8: warning: unused parameter 'chipSelectPin' [-Wunused-parameter]
C:\Users\vallim\Documents\Arduino\libraries\SdFat/SdSpiCard.h:302:8: warning: unused parameter 'sckDivisor' [-Wunused-parameter]
In file included from C:\Users\vallim\Documents\Arduino\libraries\SdFat/utility/ArduinoStream.h:29:0,
                 from C:\Users\vallim\Documents\Arduino\libraries\SdFat/utility/FatLib.h:23,
                 from C:\Users\vallim\Documents\Arduino\libraries\SdFat/SdFat.h:27,
                 from Atmega_Hex_Uploader.ino:120:
C:\Users\vallim\Documents\Arduino\libraries\SdFat/utility/bufstream.h:67:8: warning: unused parameter 'off' [-Wunused-parameter]
   bool seekoff(off_type off, seekdir way) {
        ^
C:\Users\vallim\Documents\Arduino\libraries\SdFat/utility/bufstream.h:67:8: warning: unused parameter 'way' [-Wunused-parameter]
C:\Users\vallim\Documents\Arduino\libraries\SdFat/utility/bufstream.h:139:8: warning: unused parameter 'off' [-Wunused-parameter]
   bool seekoff(off_type off, seekdir way) {
        ^
C:\Users\vallim\Documents\Arduino\libraries\SdFat/utility/bufstream.h:139:8: warning: unused parameter 'way' [-Wunused-parameter]
In file included from C:\Users\vallim\Documents\Arduino\libraries\SdFat/utility/FatLib.h:23:0,
                 from C:\Users\vallim\Documents\Arduino\libraries\SdFat/SdFat.h:27,
                 from Atmega_Hex_Uploader.ino:120:
C:\Users\vallim\Documents\Arduino\libraries\SdFat/utility/ArduinoStream.h:79:8: warning: unused parameter 'off' [-Wunused-parameter]
   bool seekoff(off_type off, seekdir way) {
        ^
C:\Users\vallim\Documents\Arduino\libraries\SdFat/utility/ArduinoStream.h:79:8: warning: unused parameter 'way' [-Wunused-parameter]
C:\Users\vallim\Documents\Arduino\libraries\SdFat/utility/ArduinoStream.h:86:8: warning: unused parameter 'pos' [-Wunused-parameter]
   bool seekpos(pos_type pos) {
        ^
C:\Users\vallim\Documents\Arduino\libraries\SdFat/utility/ArduinoStream.h:123:8: warning: unused parameter 'off' [-Wunused-parameter]
   bool seekoff(off_type off, seekdir way) {
        ^
C:\Users\vallim\Documents\Arduino\libraries\SdFat/utility/ArduinoStream.h:123:8: warning: unused parameter 'way' [-Wunused-parameter]
C:\Users\vallim\Documents\Arduino\libraries\SdFat/utility/ArduinoStream.h:126:8: warning: unused parameter 'pos' [-Wunused-parameter]
   bool seekpos(pos_type pos) {
        ^
In file included from Atmega_Hex_Uploader.ino:131:0:
Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
   };  // end of signatures
   ^
Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
Signatures.h:113:3: warning: missing initializer for member 'signatureType::timedWrites' [-Wmissing-field-initializers]
[... cut ...]
File_Utils:325: error: 'class SdFile' has no member named 'getFilename'
Programming_Utils.ino: In function 'void showHex(byte, boolean, boolean)':
Programming_Utils.ino:66:38: warning: narrowing conversion of '((((int)(b >> 4)) & 255) | 48)' from 'int' to 'char' inside { } is ill-formed in C++11 [-Wnarrowing]
Programming_Utils.ino:66:56: warning: narrowing conversion of '((((int)b) & 15) | 48)' from 'int' to 'char' inside { } is ill-formed in C++11 [-Wnarrowing]
'class SdFile' has no member named 'getFilename'

nickgammon, 2015-09-17:
Can you make the errors "code" format by putting them into backticks?

```
    ```
    Like this.
    ```
```

Please post how you configured your defines at the start of the main sketch so I can try to reproduce this. For example:

```C++
const bool allowTargetToRun = true;  // if true, programming lines are freed when not programming

#define ALLOW_MODIFY_FUSES true   // make false if this sketch doesn't fit into memory
#define ALLOW_FILE_SAVING true    // make false if this sketch doesn't fit into memory
#define SAFETY_CHECKS true        // check for disabling SPIEN, or enabling RSTDISBL

#define USE_ETHERNET_SHIELD false  // Use the Arduino Ethernet Shield for the SD card

// make true if you have spare pins for the SD card interface
#define SD_CARD_ACTIVE true

// make true to use the high-voltage parallel wiring
#define HIGH_VOLTAGE_PARALLEL false
// make true to use the high-voltage serial wiring
#define HIGH_VOLTAGE_SERIAL false
// make true to use ICSP programming
#define ICSP_PROGRAMMING true

// make true to use bit-banged SPI for programming
#define USE_BIT_BANGED_SPI true
```

mythral0, 2015-09-17:
Hi,
Problem solved, I feel quite silly, It works perfect now.
I wasn't setting the consts, was just using the ones you provide default.
Still don't really understand why it wouldn't compile though.

Many thanks for all your help.

Kind Regards

Mo.

closed 2015-09-17 by mythral0

hhumbertoAv, 2016-01-28:
Hi...How did dou solve the getFilename() problem?

Thanks a lot!

nickgammon, 2016-01-28:
How do you have the #defines set up? (see a couple of posts further up)

hhumbertoAv, 2016-01-29:
Yes...The "default defines" are the same as you have posted now...Im using My own make file and I get the same error...An other issue: The function getFilename() doesnt exist in the SdFat.h file.... :(

nickgammon, 2016-01-31:
As I mentioned on the supporting web page, I used an older version of the SDfat library, because changes they made recently seemed to break this sketch.

hhumbertoAv, 2016-02-03:
Thanks for your help...Im really worried about my job. I have solved the sd card problem using the  tinyFat Lib. However, I cant write in a micro yet. Im trying to write from an arduino mega to an atmega128 microcontroller. In the signatures I only add:   `"{ { 0x1E, 0x97, 0x02 }, "ATmega128A",  128 * kb, 1 * kb,   128, NO_FUSE }" `line. Do you thing its wrong?.... :(

nickgammon, 2016-02-03:
The datasheet says the page size is 128 words, so that is 256 bytes.

hhumbertoAv, 2016-02-03:
I ve tried it too...but it doesnt work :(...  `{ { 0x1E, 0x97, 0x02 }, "ATmega128A",  128 * kb, 4 * kb,   256, NO_FUSE  }, ` ..NO_FUSE is correct here?

nickgammon, 2016-02-03:
Can you start a thread on the Arduino forum please? I don't want to do debugging of issues in multiple places. You can send me a PM with the forum thread URL. Please clarify what "it doesn't work" means.

hhumbertoAv, 2016-02-06:
Nick...Thanks for all!!!!!...I could compile the code in the correct way and it works!!!!!...There was a problem with the compiler stack I guess....The content of the function chooseInputFile() was pasted in the  write flashcontents() function and it works!!!!

Now, is it possible to change the pins software vía?....I cant connect the atmega128 in the 4,5,6,7 arduino pins because those are used by a TFT screen....I moved atmega 128 to  the 8 9 10 11 pins, but i dont know how to change it in the scketch!

nickgammon, 2016-02-06:
Yes it is possible. There are some defines and consts near the start of the sketch. Identify which ones you need and change those.