nickgammon/arduino_sketches #27: Download

ooanea, 2015-10-20:
Hi,

I just replaced my arduino atmega16U2 and I need to reprogram it with bootloader then the firmware. I found interesting information here but how can I download the Atmega_Board_Programmer folder? 

Can someone help, and then I need to place in my software I believe but where.  A lot of esential basic information is missing.

thanks, Octavian

nickgammon, 2015-10-21:
There is a "Download ZIP" button on the main page: https://github.com/nickgammon/arduino_sketches

There should be useful information here: http://www.gammon.com.au/bootloader

ooanea, 2015-10-22:
Dear Nick,

I have managed to download, follow your instructions and brought to life my
dead Arduino. I want to thank you for sharing the long hours work I
believe. Impressive work.

Take care, Octavian,

nickgammon, 2015-10-22:
Glad I could help.

closed 2015-10-22 by nickgammon