nickgammon/arduino_sketches #28: Does Atmega_Board_Programmer support ATMEGA 644 ?

bilics, 2015-10-30:
I see that the Board Detector has it - wondering if the bootloader programmer has it.

Thanks!

nickgammon, 2015-10-30:
It looks like there is no bootloader there for that chip. (Each one is in a separate .h file). The web page <http://www.gammon.com.au/bootloader> describes how you can make a new .h file for other bootloaders.

bilics, 2015-10-30:
Thanks - I will take a look and get back here if I have any doubts.

bilics, 2015-10-31:
Hi Nick - where can I find the .hex file for the atmega 644 ?

nickgammon, 2015-10-31:
I'm not sure where the official one is. I found this thread: http://forum.arduino.cc/index.php?topic=45494.0

You could ask on the Arduino Forum in the "Microcontrollers" section, perhaps referencing that thread, and asking where the latest bootloader is. I think I last got the "Sanguino" one from memory.

closed 2018-04-03 by nickgammon