nickgammon/arduino_sketches #29: Compilation error

Znikos, 2015-11-12:
Hello guys,
I tried to compil the atmega_board_detector with a 1.6.5 arduino IDE. I have the following errors and I don't understand...
Thanks for your help!

Atmega_Board_Detector\md5.c.o: In function `md5_starts':
C:\Users\pvack\Documents\Arduino\libraries\Atmega_Board_Detector/md5.c:40: multiple definition of `md5_starts'
md5.c.o:C:\Users\pvack\AppData\Local\Temp\build5762215785007529461.tmp/md5.c:40: first defined here
Atmega_Board_Detector\md5.c.o: In function `md5_starts':
C:\Users\pvack\Documents\Arduino\libraries\Atmega_Board_Detector/md5.c:40: multiple definition of `md5_process'
md5.c.o:C:\Users\pvack\AppData\Local\Temp\build5762215785007529461.tmp/md5.c:40: first defined here
Atmega_Board_Detector\md5.c.o: In function `md5_starts':
C:\Users\pvack\Documents\Arduino\libraries\Atmega_Board_Detector/md5.c:40: multiple definition of `md5_update'
md5.c.o:C:\Users\pvack\AppData\Local\Temp\build5762215785007529461.tmp/md5.c:40: first defined here
Atmega_Board_Detector\md5.c.o: In function `md5_starts':
C:\Users\pvack\Documents\Arduino\libraries\Atmega_Board_Detector/md5.c:40: multiple definition of `md5_finish'
md5.c.o:C:\Users\pvack\AppData\Local\Temp\build5762215785007529461.tmp/md5.c:40: first defined here
collect2.exe: error: ld returned 1 exit status
Erreur lors de la compilation.

nickgammon, 2015-11-12:
I get no error compiling under IDE 1.6.5. Have you changed any of the defines from the downloaded version?

```
C:\Users\pvack\Documents\Arduino\libraries\Atmega_Board_Detector/md5.c
```

The code is **not** a library. It should be installed in your sketchbook folder like a normal sketch.

Znikos, 2015-11-20:
TY Nick!
I sorry with this question of newbie...
It works!

closed 2015-11-20 by Znikos