nickgammon/arduino_sketches #30: Problem with LilyPad bootloader

billinka, 2016-01-27:
Hello Nick,

I'm having trouble burning the bootloader to an ATMEGA328P for using the internal 8mhz clock. I am trying to make the 'minimal configuration' with the internal 8Mhz clock, from : https://www.arduino.cc/en/Tutorial/ArduinoToBreadboard

Using your Atmega Board Programmer, the bootloader burns fine, but I am unable to program the chip afterwards. Note: Reburning it as an Uno, I am able to program it.

The next time I went to burn the Lilypad bootloader I used the 'Verify' option and got a lot of verification errors:

'''
Type 'L' to use Lilypad (8 MHz) loader, or 'U' for Uno (16 MHz) loader ...
Using Lilypad 8 MHz loader.
Bootloader address = 0x7800
Bootloader length = 1932 bytes.
Type 'Q' to quit, 'V' to verify, or 'G' to program the chip with the bootloader ...
Verifying ...
Verification error at address 7800. Got: 0xFF  Expected: 0x0C 
Verification error at address 7801. Got: 0xFF  Expected: 0x94 
Verification error at address 7802. Got: 0xFF  Expected: 0x34 
Verification error at address 7803. Got: 0xFF  Expected: 0x3C 
Verification error at address 7804. Got: 0xFF  Expected: 0x0C 
Verification error at address 7805. Got: 0xFF  Expected: 0x94 
Verification error at address 7806. Got: 0xFF  Expected: 0x51 
Verification error at address 7807. Got: 0xFF  Expected: 0x3C 
Verification error at address 7808. Got: 0xFF  Expected: 0x0C 
Verification error at address 7809. Got: 0xFF  Expected: 0x94 
Verification error at address 780A. Got: 0xFF  Expected: 0x51 
Verification error at address 780B. Got: 0xFF  Expected: 0x3C 
Verification error at address 780C. Got: 0xFF  Expected: 0x0C 
Verification error at address 780D. Got: 0xFF  Expected: 0x94 
Verification error at address 780E. Got: 0xFF  Expected: 0x51 
Verification error at address 780F. Got: 0xFF  Expected: 0x3C 
Verification error at address 7810. Got: 0xFF  Expected: 0x0C 
Verification error at address 7811. Got: 0xFF  Expected: 0x94 
Verification error at address 7812. Got: 0xFF  Expected: 0x51 
Verification error at address 7813. Got: 0xFF  Expected: 0x3C 
Verification error at address 7814. Got: 0xFF  Expected: 0x0C 
Verification error at address 7815. Got: 0xFF  Expected: 0x94 
Verification error at address 7816. Got: 0xFF  Expected: 0x51 
Verification error at address 7817. Got: 0xFF  Expected: 0x3C 
Verification error at address 7818. Got: 0xFF  Expected: 0x0C 
Verification error at address 7819. Got: 0xFF  Expected: 0x94 
Verification error at address 781A. Got: 0xFF  Expected: 0x51 
Verification error at address 781B. Got: 0xFF  Expected: 0x3C 
Verification error at address 781C. Got: 0xFF  Expected: 0x0C 
Verification error at address 781D. Got: 0xFF  Expected: 0x94 
Verification error at address 781E. Got: 0xFF  Expected: 0x51 
Verification error at address 781F. Got: 0xFF  Expected: 0x3C 
Verification error at address 7820. Got: 0xFF  Expected: 0x0C 
Verification error at address 7821. Got: 0xFF  Expected: 0x94 
Verification error at address 7822. Got: 0xFF  Expected: 0x51 
Verification error at address 7823. Got: 0xFF  Expected: 0x3C 
Verification error at address 7824. Got: 0xFF  Expected: 0x0C 
Verification error at address 7825. Got: 0xFF  Expected: 0x94 
Verification error at address 7826. Got: 0xFF  Expected: 0x51 
Verification error at address 7827. Got: 0xFF  Expected: 0x3C 
Verification error at address 7828. Got: 0xFF  Expected: 0x0C 
Verification error at address 7829. Got: 0xFF  Expected: 0x94 
Verification error at address 782A. Got: 0xFF  Expected: 0x51 
Verification error at address 782B. Got: 0xFF  Expected: 0x3C 
Verification error at address 782C. Got: 0xFF  Expected: 0x0C 
Verification error at address 782D. Got: 0xFF  Expected: 0x94 
Verification error at address 782E. Got: 0xFF  Expected: 0x51 
Verification error at address 782F. Got: 0xFF  Expected: 0x3C 
Verification error at address 7830. Got: 0xFF  Expected: 0x0C 
Verification error at address 7831. Got: 0xFF  Expected: 0x94 
Verification error at address 7832. Got: 0xFF  Expected: 0x51 
Verification error at address 7833. Got: 0xFF  Expected: 0x3C 
Verification error at address 7834. Got: 0xFF  Expected: 0x0C 
Verification error at address 7835. Got: 0xFF  Expected: 0x94 
Verification error at address 7836. Got: 0xFF  Expected: 0x51 
Verification error at address 7837. Got: 0xFF  Expected: 0x3C 
Verification error at address 7838. Got: 0xFF  Expected: 0x0C 
Verification error at address 7839. Got: 0xFF  Expected: 0x94 
Verification error at address 783A. Got: 0xFF  Expected: 0x51 
Verification error at address 783B. Got: 0xFF  Expected: 0x3C 
Verification error at address 783C. Got: 0xFF  Expected: 0x0C 
Verification error at address 783D. Got: 0xFF  Expected: 0x94 
Verification error at address 783E. Got: 0xFF  Expected: 0x51 
Verification error at address 783F. Got: 0xFF  Expected: 0x3C 
Verification error at address 7840. Got: 0xFF  Expected: 0x0C 
Verification error at address 7841. Got: 0xFF  Expected: 0x94 
Verification error at address 7842. Got: 0xFF  Expected: 0x51 
Verification error at address 7843. Got: 0xFF  Expected: 0x3C 
Verification error at address 7844. Got: 0xFF  Expected: 0x0C 
Verification error at address 7845. Got: 0xFF  Expected: 0x94 
Verification error at address 7846. Got: 0xFF  Expected: 0x51 
Verification error at address 7847. Got: 0xFF  Expected: 0x3C 
Verification error at address 7848. Got: 0xFF  Expected: 0x0C 
Verification error at address 7849. Got: 0xFF  Expected: 0x94 
Verification error at address 784A. Got: 0xFF  Expected: 0x51 
Verification error at address 784B. Got: 0xFF  Expected: 0x3C 
Verification error at address 784C. Got: 0xFF  Expected: 0x0C 
Verification error at address 784D. Got: 0xFF  Expected: 0x94 
Verification error at address 784E. Got: 0xFF  Expected: 0x51 
Verification error at address 784F. Got: 0xFF  Expected: 0x3C 
Verification error at address 7850. Got: 0xFF  Expected: 0x0C 
Verification error at address 7851. Got: 0xFF  Expected: 0x94 
Verification error at address 7852. Got: 0xFF  Expected: 0x51 
Verification error at address 7853. Got: 0xFF  Expected: 0x3C 
Verification error at address 7854. Got: 0xFF  Expected: 0x0C 
Verification error at address 7855. Got: 0xFF  Expected: 0x94 
Verification error at address 7856. Got: 0xFF  Expected: 0x51 
Verification error at address 7857. Got: 0xFF  Expected: 0x3C 
Verification error at address 7858. Got: 0xFF  Expected: 0x0C 
Verification error at address 7859. Got: 0xFF  Expected: 0x94 
Verification error at address 785A. Got: 0xFF  Expected: 0x51 
Verification error at address 785B. Got: 0xFF  Expected: 0x3C 
Verification error at address 785C. Got: 0xFF  Expected: 0x0C 
Verification error at address 785D. Got: 0xFF  Expected: 0x94 
Verification error at address 785E. Got: 0xFF  Expected: 0x51 
Verification error at address 785F. Got: 0xFF  Expected: 0x3C 
Verification error at address 7860. Got: 0xFF  Expected: 0x0C 
Verification error at address 7861. Got: 0xFF  Expected: 0x94 
Verification error at address 7862. Got: 0xFF  Expected: 0x51 
Verification error at address 7863. Got: 0xFF  Expected: 0x3C 
Verification error at address 7864. Got: 0xFF  Expected: 0x0C 
1889 verification error(s).
First 100 shown.
Programming mode off.
'''

nickgammon, 2016-01-27:
It looks like there is nothing there (all 0xFF). Rather than having a lengthy chat about it here, can you open a thread on the Arduino forum, and ask this? Send me a link to it so I  notice it.

billinka, 2016-01-27:
It verifies fine if I select the 'U' option.

Here is a link to my post on Arduino Forum:
http://forum.arduino.cc/index.php?topic=374999.0

closed 2016-01-27 by billinka