nickgammon/arduino_sketches #32: Atmega Board Programmer for teensy 2.0

bhaap, 2016-04-08:
Hi

Im trying to get your programmer to work on my teensy 2.0. 
Im able to build it and upload it to my Teensy but after that its not going so well. 
The teensy is not showing up as a serial device so i am unable to enter serial monitor to start burning bootloaders. 

Can you point me in the right direction to be able to use my teensy with your program?

nickgammon, 2016-04-09:
Can you get the Teensy to show up in the serial monitor using *any* program? That would be the first step. I think they have example code for using the serial monitor.

bhaap, 2016-04-09:
Yes I have a modified ArduinoISP for example and that works flawless :P Just that im having trouble with the Mega2650's 256k memory. (I think the ArduinoISP is limited to 64k or so). Ive also run the Teensy as a PS2 to USB adapter and a ADB to USB adapter aswell.

The Teensy does show up in the system info, but as a "composite device" what ever that means. (Im running Mac as a sidenote)

nickgammon, 2016-04-10:
> I think the ArduinoISP is limited to 64k or so

Yes I think that was one of the issues with it.

> The Teensy does show up in the system info, but as a "composite device" what ever that means.

Hmm - you need to get a standard communications sketch to work like the ASCIItable one. Until you do that, fiddling with this sketch won't achieve much.

bhaap, 2016-04-11:
the ASCIItable sketch works fine. https://www.arduino.cc/en/Tutorial/ASCIITable

I get a serial connection and the correct ASCIItable output from the serial connection.

bhaap, 2016-04-11:
Video showing what i mean.

https://www.youtube.com/watch?v=TVf6J7Z5tQ4

nickgammon, 2016-04-12:
Very nice! I don't have a Teensy 2.0, but I do have a Teensy 3.0. I had to comment out the following lines from the initPins function to get it to compile:

```
 // set up 8 MHz timer on pin 9
  pinMode (CLOCKOUT, OUTPUT);
  // set up Timer 1
  TCCR1A = bit (COM1A0);  // toggle OC1A on Compare Match
  TCCR1B = bit (WGM12) | bit (CS10);   // CTC, no prescaling
  OCR1A =  0;       // output every cycle
```

However once I did that it uploaded OK, and I could open the serial monitor:

```
Atmega chip programmer.
Written by Nick Gammon.
Version 1.36
Compiled on Apr 12 2016 at 15:37:21 with Arduino IDE 10605.
Attempting to enter ICSP programming mode ......................................................
Failed to enter programming mode. Double-check wiring!
Type 'C' when ready to continue with another chip ...

```

I can't explain why the Teensy 2.0 doesn't work.

bhaap, 2016-04-12:
Oh, didnt even know that the 3.0 worked with arduino sketches. I thought it was a 32-bit ARM chip on that one. I think i have a Teensy 3.1 somewhere, might try that one instead.

bhaap, 2016-04-12:
Also, the Atmega_board_detector sketch works for some reason.

bhaap, 2016-04-12:
Hmm might have been that it took to much memory on my teensy. when i changed the first lines to this it worked:

`
#define USE_ATMEGA8 false
#define USE_ATMEGA16U2 true    // Uno USB interface chip
#define USE_ATMEGA32U4 true    // Leonardo
#define USE_ATMEGA168 false
#define USE_ATMEGA1280 false
#define USE_ATMEGA1284 false
#define USE_ATMEGA2560 true
#define USE_ATMEGA256RFR2 false // Pinoccio Scout`

nickgammon, 2016-04-12:
>  I thought it was a 32-bit ARM chip on that one.

It is, but it can still program an AVR chip. The programming sequences can still be generated.

closed 2018-04-03 by nickgammon