nickgammon/arduino_sketches #33: make unique ID with chip specification

mkeyno, 2016-04-13:
hi @nickgammon , I have plan to protect my sketch by similar  unique MAC address, so in setup function check whether sketch's license  match the chip ID or not and if ok goto loop function 
 storing the address in EEPROM is not good idea because every body can extract hex code of both flash and EEPROM and use it in other hardware , I really appreciated if you could help me 
BR

nickgammon, 2016-04-14:
As I have made these sketches open source, I have no motivation to help people protect them by implementing copy-protection. If you are referring to something other than my sketches then I suggest posting a question on the Arduino Forum or the Arduino Stack Exchange.

mkeyno, 2016-04-14:
ok , thanks anyway , I just was wondering whether your technique could be use for such protection  or  I should find another way

closed 2018-04-03 by nickgammon