nickgammon/arduino_sketches #34: Support to change onboard LED pin function

csavy, 2016-06-06:
I have a custom MEGA2560-based board that uses D10 instead of D13 for an onboard status LED.

I understand that the optiboot bootloader code flashed D13:
-at Arduino boot time (the LED blinks a few times)
-when uploading a sketch to Arduino

Please allow the user to change the pin associated with the functionality of the LED, preferably in the 
file ATmega_Board_Programmer.ino. I do not feel experienced enough to re-compile the bootloader on my own!

P.S. your code is awesome, what a life saver for custom projects - especially on MEGA2560!

nickgammon, 2016-06-06:
The bootloaders are just hard-coded as precompiled binaries. I don't have the ability to change that in the way you request.

csavy, 2016-06-07:
Thank you for the response. May I ask if you provide .C files I can edit and compile to institute this change?

If not, is there another way to implement his change? Perhaps links to some more tutorials like yours?

-Chris

Sent from my iPhone. Please excuse any typos.

csavy, 2016-06-07:
> You would need to recompile the appropriate bootloader. I think the bootloader sources are provided with the Arduino download, and if not are on the Arduino Github site. Recompiling the bootloader can be a bit non-trivial if you aren't used to it, because the IDE doesn't do it by default. See here for possible help: http://arduino.stackexchange.com/questions/23900/atmega328p-sck-pin/23914#23914
>
> -

Much appreciated Nick, thanks again!

-Chris

closed 2018-04-03 by nickgammon