nickgammon/arduino_sketches #35: error while compling

amirezz, 2016-07-07:
dear nickgammon
thanks for that great job and for sharing it
i have this error while compling
Using library SdFat in folder: C:\Users\miro\Documents\Arduino\libraries\SdFat (legacy)
exit status 1
'class SdFile' has no member named 'getFilename'

any ideas

gknauf, 2016-09-26:
Hi Nick,
I can only second amirezz's issue;
seems that the SdFat library was heavily changed in July, can you probably for now share the SdFat version you used (upload to GoogleDrive or something like that) since SdFat does not offer any older releases for download ...

nickgammon, 2016-09-26:
I mentioned this issue on my page here: http://www.gammon.com.au/uploader

> The SdFat library seems to be undergoing changes/improvements as at April 2015. The hex uploader may not compile with the new version of the SdFat library.
>
> An archived version of the SdFat library (which works with the hex uploader sketch) is available from:
> 
> http://gammon.com.au/Arduino/SdFat-master.zip (2.2 Mb)
>
> Unzip that file, and from within the folders inside it, copy the SdFat folder into your "libraries" folder (which is under your Arduino sketchbook folder - not inside the Arduino application folder). Then restart the Arduino IDE.

gknauf, 2016-09-26:
Thanks Nick, that works for me; Atmega_Hex_Uploader compiles fine with this SdFat library version and displays the files on my SD card ...
Though I wonder whats missing with the SD library that comes with the IDE and why you dont use that one instead ...

nickgammon, 2016-09-26:
I don't know. I tried to make it work once and decided that the effort involved didn't really have any useful pay-off.

observative, 2017-12-01:
Thank you for dually posting the SdFat files, I missed the note on the forum page first round.

nickgammon, 2017-12-02:
Glad it was helpful. I'll close this issue now as it seems to be resolved.

closed 2017-12-02 by nickgammon