nickgammon/arduino_sketches #37: Bootloader versions

PRosenb, 2016-08-14:
Dear nickgammon,
Many thinks for creating these sketches, they helped me much.
I'm wondering where I can see the versions of the bootloaders used. Could you please add them if they are not mentioned somewhere? I think perfect would be to mention them in the corresponding header files like bootloader_atmega328.h.

nickgammon, 2016-08-20:
How would you define the bootloader version? I think the same bootloader is used from one Arduino release to the next, unless they find a major bug.

PRosenb, 2016-08-20:
Good point. An option would be to mention the Git commit hash on Github (or the link to it). Are you using the [optiboot](https://github.com/Optiboot/) bootloader?
I burned the 8Mhz boatloader to the ATmega328 and experience problems with the watchdog reset. Instead of doing a reset, it gets stuck with a flashing LED.
The behaviour seams to be like on old bootloaders.

per1234, 2016-08-20:
Optiboot does have a version: https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/bootloaders/optiboot/optiboot.c#L146-L147
Some of the other bootloaders included in Arduino AVR Boards have a change log date but that relies on the change log being updated on every edit so the commit hash of the last modification of the file seems most reliable. For example, if you downloaded ATmegaBOOT_168_atmega328_pro_8MHz.hex from the Arduino IDE repository right now I would add this comment to bootloader_lilypad328.h:
```
// https://github.com/arduino/Arduino/blob/1b3ae5fa63c2ca7c5a585f3929d382b441bbba9f/hardware/arduino/avr/bootloaders/atmega/ATmegaBOOT_168_atmega328_pro_8MHz.hex
```
That will make it clear where the file came from and make it possible for users to check if there have been any significant changes since the version you include with Atmega_Board_Programmer. By looking at the history for the file: https://github.com/arduino/Arduino/commits/master/hardware/arduino/avr/bootloaders/atmega/ATmegaBOOT_168_atmega328_pro_8MHz.hex you can see there have been no commits since.

nickgammon, 2016-08-31:
As best I could I added more version information to the .h files. For the ones where I could locate the originals I also include the .hex file MD5 sums, so you can quickly check if it is the same file as the current version.

PRosenb, 2016-09-01:
That would be great.

closed 2016-09-03 by PRosenb