nickgammon/arduino_sketches #38: Optiboot

PRosenb, 2016-09-03:
I converted the optiboot bootloader [optiboot_atmega328_pro_8MHz](https://github.com/Optiboot/optiboot/blob/master/optiboot/bootloaders/optiboot/optiboot_atmega328_pro_8MHz.hex) with convertHexToByteArray.py, replaced the file bootloader_lilypad328.h in Atmega_Board_Programmer with it and adjusted the start address in Atmega_Board_programmer on line 418 to 0x7e00.
Burning and verifying works afterwards but I am not able to upload a sketch to it using the 8Mhz breadboard board in the Arduino IDE. The LED on output 13 flashes three times quickly.
Any hints are appreciated.

nickgammon, 2016-09-23:
Did you also change the fuse?

Original:

```
      newhFuse = 0xDA;  //  2048 byte bootloader, SPI enabled
      addr = 0x7800;
```

I suggest omitting the line which changes the fuse (newhFuse = 0xDA; ) because the original would have used the 512 byte bootloader.

PRosenb, 2016-09-24:
Do you mean just uncomment the newhFuse line?

I tried both without luck:
```c++
      newlFuse = 0xE2;  // internal 8 MHz oscillator
//      newhFuse = 0xDA;  //  2048 byte bootloader, SPI enabled
      addr = 0x7e00;
```
and
```c
      newlFuse = 0xE2;  // internal 8 MHz oscillator  
      newhFuse = 0xDA;  //  2048 byte bootloader, SPI enabled  
      addr = 0x7e00;  
```

nickgammon, 2016-09-24:
The newhFuse line should be commented out or removed (in your case) because you want the same bootloader size as the "normal" board (the Optiboot one). Can you upload the bootloader with the line commented out (removed), and then run my "board detector" sketch to confirm what your fuse settings are?

PRosenb, 2016-09-24:
Thanks for the quick reply.
I commented it out, programmed it and then run the "board detector" sketch with the following output
```
LFuse = 0xE2 
HFuse = 0xDE 
EFuse = 0xFD 
Lock byte = 0xEF 
Clock calibration = 0xA5 
Bootloader in use: Yes
EEPROM preserved through erase: No
Watchdog timer always on: No
Bootloader is 512 bytes starting at 7E00
```

nickgammon, 2016-09-24:
That looks correct. The fact that the LED on output 13 flashes 3 times also indicates the bootloader is working. Now the issue is why it doesn't upload. This isn't really an issue with this code. What board do you have selected in the IDE? It is probably sending commands at the wrong baud rate or using the wrong protocol.

PRosenb, 2016-09-27:
I tried many different boards and changed boards.txt to the baud rates `57600` and `115200`, tried the protocols `arduino` and `stk500` and followed instructions of various forums. All without any luck. What configuration is it supposed to use?

PRosenb, 2016-12-22:
@NicoHood I saw that you created the py script to convert the bootloaders to hex files. Have you managed to convert an 8MHz Optiboot bootloader for AVR 328 as well?

NicoHood, 2016-12-23:
@PRosenb I did not try, but how would it matter? Its just different bytecode, but the script will not work different.

PRosenb, 2017-01-05:
@NicoHood I tried it as described in this thread but had no success. Maybe it was because of the start address but I have no hardware to find it out.

closed 2018-04-03 by nickgammon