nickgammon/arduino_sketches #39: Atmega_Self_Read_Signature not reading bootloader

gknauf, 2016-09-11:
Hi Nick, 
thanks for making your great programs available!
I tested Atmega_Self_Read_Signature on some of my boards, but only with my China Uno clone the sketch dumps something which looks like a real bootloader, with my China Nano and 2 China ProMini I get only:

Bootloader:

7800: 02 0F 0F 0F 0F 0F 0F 0F 0F 0F 0F 0F 0F 0F 0F 0F 
7810: 02 0F 0F 0F 0F 0F 0F 0F 0F 0F 0F 0F 0F 0F 0F 0F 
...

as you wrote at other places this is most likely because of the lock fuse (0xCF); but wouldnt it be better to just write out "bootloader is locked and cant be read" instead of dumping the garbage?

nickgammon, 2016-09-23:
You could be right, but a bad read-back could be for lots of reasons. Are you suggesting checking the lock byte?

gknauf, 2016-09-26:
Hi Nick,
basically yes. But meanwhile I found another issue with IDE 1.6.9: the gcc-internaL memcpy clashes with the one you intent to use from string.h; so I added:
#pragma GCC optimize ("-O0") // avoid GCC memcpy inline
anf now I get the same bootloader output as you wrote in your forum:
Bootloader:

7800: 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 
7810: 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 
7820: 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 

also I added some signatures, so the full patch is:
```C
--- Atmega_Self_Read_Signature.ino.orig	2016-08-31 07:31:47 +0200
+++ Atmega_Self_Read_Signature.ino	2016-09-18 19:51:30 +0200
@@ -31,6 +31,7 @@
  or the use or other dealings in the software. 
 
 */
+#pragma GCC optimize ("-O0") // avoid GCC memcpy inline
 
 #include <avr/boot.h>
 #include <avr/pgmspace.h>
@@ -78,7 +79,9 @@
   { { 0x1E, 0x92, 0x0A }, "ATmega48PA",   4 * kb,         0 },
   { { 0x1E, 0x93, 0x0F }, "ATmega88PA",   8 * kb,       256 },
   { { 0x1E, 0x94, 0x0B }, "ATmega168PA", 16 * kb,       256 },
+  { { 0x1E, 0x94, 0x06 }, "ATmega168V",  16 * kb,       256 },
   { { 0x1E, 0x95, 0x0F }, "ATmega328P",  32 * kb,       512 },
+  { { 0x1E, 0x95, 0x16 }, "ATmega328PB", 32 * kb,       512 },
 
   // Atmega644 family
   { { 0x1E, 0x94, 0x0A }, "ATmega164P",   16 * kb,      256 },
@@ -107,6 +110,7 @@
   
   // ATmega1284P family
   { { 0x1E, 0x97, 0x05 }, "ATmega1284P", 128 * kb,   1 * kb },
+  { { 0x1E, 0x97, 0x06 }, "ATmega1284",  128 * kb,   1 * kb },
   
   // ATtiny4313 family
   { { 0x1E, 0x91, 0x0A }, "ATtiny2313A", 2 * kb,   0 },
##############################################################################
--- md5.c.orig	2016-08-31 07:31:47 +0200
+++ md5.c	2016-09-12 17:01:34 +0200
@@ -17,7 +17,9 @@
  *  along with this program; if not, write to the Free Software
  *  Foundation, Inc., 59 Temple Place, Suite 330, Boston, MA  02111-1307  USA
  */
+#pragma GCC optimize ("-O0") // avoid GCC memcpy inline
 
+#include <string.h>  // for memcpy
 #include "md5.h"
 
 #define GET_UINT32(n,b,i)                       \
```
I think it would also make sense to #ifdef the bunch of signatures coming after ATmega168 types or else the sketch becomes too big to fit into the 16k ...

I have a similar patch for Atmega_Board_Detector which I will post with a new issue.

[Atmega_Self_Read_Signature.diff.txt](https://github.com/nickgammon/arduino_sketches/files/492012/Atmega_Self_Read_Signature.diff.txt)

nickgammon, 2016-11-18:
See https://github.com/nickgammon/arduino_sketches/commit/5aab29acd088d2bd853ed7234b609af3babe98f5

I didn't do the #ifdef, I might leave that as an exercise. :)

closed 2018-04-03 by nickgammon