nickgammon/arduino_sketches #41: Use bootloader

judavi, 2016-10-10:
Hi! Sorry I'm really n00b in this topics so I would like to ask how I use the bootlader for a Lilypad USB? I mean where I run the code or how I use the code? Thanks so much for all help that you could bring me.

nickgammon, 2016-10-10:
The Lilypad USB should have come with a bootloader. I wouldn't be attempting to upload a new one unless you have a pressing reason to.

If you have a general question about how to use the Lilypad I suggest you post a question on the [Arduino Forum](http://forum.arduino.cc/)

judavi, 2016-10-11:
My issue it's because one of two that I bought comes without the bootloader. I get this conclusion connecting both on different computers and the second one it's never recognized.  I will post my question there. Thanks so much :)

closed 2018-04-03 by nickgammon