nickgammon/arduino_sketches #42: Board Detector: Add Attiny Bootloader Signatures

prologician, 2016-11-05:
Hello, @nickgammon . Firstly, thank you for your tools provided here. I've used the Atmega_Board_Detector and Atmega_Self_Read_Signature several times in order to get insights into what bootloaders my Arduinos are running.

As an addition to the existing library of bootloader hashes included in the board detector, it would be great if Attiny bootloaders were included, particularly the versions of micronucleus (used in Digispark and Digispark Pro), the Adafruit Trinket bootloader, and the Arduino Gemma bootloader.
* https://github.com/micronucleus/micronucleus
* https://github.com/adafruit/Adafruit-Trinket-Gemma-Bootloader
* The Arduino Gemma bootloader looks to be included with the recent versions of the Arduino IDE. It claims to be the same as the Adafruit one, but using different USB VID/PID values.

(I've done some brief research on this, and it looks like there isn't going to be a definitive answer on how to do this, because there is no specialized "bootloader space" on these devices. Even so, a heuristic would be nice... in the case of Attiny85, potentially hash the memory from 0x1C00 to 0x1FFF and use that as the signature basis? Reading the reset vector might help in determining the bootloader size, but I don't know how to even do that.)

nickgammon, 2016-11-18:
I think those "bootloaders" are just code that runs in normal mode that can simulate the way other bootloaders work (in other words, they are not a protected part of PROGMEM). I suppose you could add a signature for them, if we wanted to check a specific memory address range to contain a certain checksum.

nickgammon, 2018-04-03:
I'm closing this because this hardware does not technically have a bootloader, and it would be tedious to check bootloader signatures in some random part of PROGMEM, particularly if the developers of those bootloaders happen to keep updating them, so the signatures would change.

closed 2018-04-03 by nickgammon