nickgammon/arduino_sketches #43: Verification error at address 7861. Got: 0x30  Expected: 0x94

sineverba, 2016-12-21:
Hi to all, and thank you for this fantastic library!
First of all, I want say that with the "official guide" Arduino To Breadobard on the official site of Arduino every time I get the error "Yikes! Invalide signature".

With Gammon detector, I can read that I have a ATMEGA328P on breadboard, programmed with a bootloader that use the ext crystal.

I have only 2 22pf capacitator, so I did use it for the two pins current (decoupling capacitator) and I set the crystal from Pin 9.

With this configuration, as I did say, I can read and get info from the ATMEGA on breadboard (from my Arduno UNO R3 and IDE 1.6.13, last one).

But, If I load the sketches for the programmer and in serial I select Lily (to use the 8MhZ) I get 1092 errors.

Errors are:

`
[...]
Committing page starting at 0x7F80
Written.
Verifying ...
Verification error at address 7800. Got: 0x00  Expected: 0x0C 
Verification error at address 7801. Got: 0x00  Expected: 0x94 
Verification error at address 7802. Got: 0x01  Expected: 0x34 
Verification error at address 7803. Got: 0x01  Expected: 0x3C 
Verification error at address 7804. Got: 0x02  Expected: 0x0C 
Verification error at address 7805. Got: 0x02  Expected: 0x94 
Verification error at address 7806. Got: 0x03  Expected: 0x51 `

And so on.

Ok, I know that I need to buy some other capacitators and try with a real crystal (I think), but why I get that error? Another info, my breadboard get power from a USB Samsung charger (so 5 volts are not get from the UNO, to don't strees it).

Thank you

sineverba, 2016-12-21:
After connecting directly from the 5V of the Arduino... bootlader programmed. Is it possible?? Is it not very strange!?

closed 2016-12-22 by sineverba

nickgammon, 2016-12-22:
Did you connect the grounds? I always just connect the two boards together. The consumption to power another processor isn't all that great.

sineverba, 2016-12-23:
nickgammon thantk you for your response.... ground between UNO and breadboard...?
So... no.... argh! Noob fault.....