nickgammon/arduino_sketches #44: Read/Write to target board EEPROM?

DirtyEngineer, 2017-02-26:
Greetings @nickgammon, I'm really happy I found this and for all the hard work you have put in. Thank you.

First off I'm relatively new to the software side as I'm a hardware guy, so please bare with me.  How hard would it be to adapter/modify your Atmega Board Programmer code to read/write to a target boards EEPROM? 

For my specific case I'm looking to build a stand alone 328P board to read/write to the EEPROM of Atmega8's in the field without having to haul around a laptop to update them. The legacy program running on the Atmega8's reads the settings stored in it's EEPROM and runs accordingly. I have to periodically read the EEPROM from one target board and then burn it to another target board. A stand alone programmer would save a lot of time. Do you think this is an attainable goal?

nickgammon, 2017-02-27:
It's attainable, but to add this code to the current programmer would probably bloat out the code somewhat.

DirtyEngineer, 2017-02-27:
Thank you for your response. I fully understand that this would be outside of the scope of your programmer.  I was hoping to use it as a starting point. I haven't been able to find an example/project of how to to read/write from one avr to another avr's eeprom. I wasn't sure if it was possible. Your programmer was the closest progect I could find of what I'm trying to do. Any chance you could point me in the right direction?

nickgammon, 2017-02-28:
The core code in my "programmer" sketches does the complex part. You would need to look up the read/write EEPROM opcodes from the datasheet, and then write appropriate code to:

* Read from the EEPROM in case 1
* Write to the EEPROM in case 2

DirtyEngineer, 2017-03-01:
Thank you again for your insight! I really appreciate it.

Reading through the "ArduinoISP" code I think I found the opcodes

- Read from the EEPROM - "spi_transaction(0xA0, (addr >> 8) & 0xFF, addr & 0xFF, 0xFF);"
- Write to the EEPROM - "spi_transaction(0xC0, (addr >> 8) & 0xFF, addr & 0xFF, buff[x]);"

```
#define EECHUNK (32)
uint8_t write_eeprom(unsigned int length) {
  // here is a word address, get the byte address
  unsigned int start = here * 2;
  unsigned int remaining = length;
  if (length > param.eepromsize) {
    error++;
    return STK_FAILED;
  }
  while (remaining > EECHUNK) {
    write_eeprom_chunk(start, EECHUNK);
    start += EECHUNK;
    remaining -= EECHUNK;
  }
  write_eeprom_chunk(start, remaining);
  return STK_OK;
}
// write (length) bytes, (start) is a byte address
uint8_t write_eeprom_chunk(unsigned int start, unsigned int length) {
  // this writes byte-by-byte,
  // page writing may be faster (4 bytes at a time)
  fill(length);
  prog_lamp(LOW);
  for (unsigned int x = 0; x < length; x++) {
    unsigned int addr = start + x;
    spi_transaction(0xC0, (addr >> 8) & 0xFF, addr & 0xFF, buff[x]);
    delay(45);
  }
  prog_lamp(HIGH);
  return STK_OK;
}

void program_page() {
  char result = (char) STK_FAILED;
  unsigned int length = 256 * getch();
  length += getch();
  char memtype = getch();
  // flash memory @here, (length) bytes
  if (memtype == 'F') {
    write_flash(length);
    return;
  }
  if (memtype == 'E') {
    result = (char)write_eeprom(length);
    if (CRC_EOP == getch()) {
      SERIAL.print((char) STK_INSYNC);
      SERIAL.print(result);
    } else {
      error++;
      SERIAL.print((char) STK_NOSYNC);
    }
    return;
  }
  SERIAL.print((char)STK_FAILED);
  return;
}

uint8_t flash_read(uint8_t hilo, unsigned int addr) {
  return spi_transaction(0x20 + hilo * 8,
                         (addr >> 8) & 0xFF,
                         addr & 0xFF,
                         0);
}

char flash_read_page(int length) {
  for (int x = 0; x < length; x += 2) {
    uint8_t low = flash_read(LOW, here);
    SERIAL.print((char) low);
    uint8_t high = flash_read(HIGH, here);
    SERIAL.print((char) high);
    here++;
  }
  return STK_OK;
}

char eeprom_read_page(int length) {
  // here again we have a word address
  int start = here * 2;
  for (int x = 0; x < length; x++) {
    int addr = start + x;
    uint8_t ee = spi_transaction(0xA0, (addr >> 8) & 0xFF, addr & 0xFF, 0xFF);
    SERIAL.print((char) ee);
  }
  return STK_OK;
}

void read_page() {
  char result = (char)STK_FAILED;
  int length = 256 * getch();
  length += getch();
  char memtype = getch();
  if (CRC_EOP != getch()) {
    error++;
    SERIAL.print((char) STK_NOSYNC);
    return;
  }
  SERIAL.print((char) STK_INSYNC);
  if (memtype == 'F') result = flash_read_page(length);
  if (memtype == 'E') result = eeprom_read_page(length);
  SERIAL.print(result);
}
```

nickgammon, 2017-03-01:
I think we are straying here from reporting a bug in my existing code, to doing design for a different project. I suggest you make a thread on the Arduino Forum about what you are attempting.

DirtyEngineer, 2017-03-01:
Thank you for your help.

closed 2017-03-01 by DirtyEngineer