nickgammon/arduino_sketches #48: Bootloader programmer fails to enter programming mode

kxling, 2017-10-27:
Below shows what is displayed on serial monitor. what is the possible cause? my 'programming' board is arduino uno and my 'target' board is arduino pro mini 5V. i connected using manual programming cable. i successfully upload the sketch, so i doubt its the connection problem.

```
Atmega chip programmer.
Written by Nick Gammon.
Version 1.37
Compiled on Oct 27 2017 at 17:06:38 with Arduino IDE 10803.
Attempting to enter ICSP programming mode ......................................................
Failed to enter programming mode. Double-check wiring!
Type 'C' when ready to continue with another chip ...
```

nickgammon, 2017-11-01:
Not sure. Try running the chip detector sketch: http://www.gammon.com.au/forum/?id=11633

The Arduino ISP wiring and my wiring are different, I think.

nickgammon, 2018-04-03:
Any further progress? If not I'll close this issue, as probably a wiring one.

closed 2018-04-03 by nickgammon