nickgammon/arduino_sketches #50: Using EEPROM or PROGMEM for "fixed filename"

LMDIGroup, 2018-03-30:
Let me preface this by saying all of this is GREAT! I have been looking for this very functionality for a long time and I just now came across it.

What I ultimately would be interested in would be reading the .hex (or byte array of .hex) directly from PROGMEM and/or EEPROM. Do you have any plans to implement that feature? I think it would be especially useful in the 'fixed filename' sketch.

Thanks for all your doing.

nickgammon, 2018-03-31:
You are the second person to ask that in a couple of weeks. My "bootloader programmer" sketch effectively does that. With a bit of modification you could write a sketch rather than a bootloader.

LMDIGroup, 2018-03-31:
I will take a look and see what I can whip up. It does seem like a good idea especially as a 'firmware updater'. Just a smaller footprint tool you have to carry with you.

Thanks again.

nickgammon, 2018-04-01:
Your limitation would be that you need to keep in PROGMEM both the code that does the flashing, and the code you want to flash. Based on the amount of program memory needed to hold my bootloader-uploader, and removing the bootloaders from it, it looks like around 8 KB is needed (out of about 30 KB available). So, you could upload about a 22 KB sketch. This may well suit your circumstances. For more, just use the existing code that reads from an SD card.

The [fixed filename](https://github.com/nickgammon/arduino_sketches/tree/master/Atmega_Hex_Uploader_Fixed_Filename) uploader is designed to be used in the field, as you just press a switch and the designated code from the SD card is uploaded in a couple of seconds.

Crossroads, from the Arduino Forum, has made a board designed to work with that. You can select from up to 99 different files (eg. file00.hex to file99.hex) and press a switch and that file gets uploaded. Nice and simple to use in the field. And with the choice of filenames, you could take a single device that could program different boards, or boards with different options enabled.

remshaS, 2020-07-20:
is there a way i can upload a program without the switch

nickgammon, 2023-02-21:
Change the code? However when would it know to do the upload?

closed 2023-02-21 by nickgammon