nickgammon/arduino_sketches #51: ATmega328PB not programming

ryanoc3ros, 2018-04-02:
I am attempting to burn a bootloader on a ATmega328PB. I see version 1.38 added support for this chip. When connecting a brand new chip to a master arduino with Atmega_Board_Programmer.ino, it detects the chip, and prompts me to start programming the board. It gets to the "Divide clock by 8" prompt and changes a fuse. After this, the program can no longer find the board. It just says "Attempting to enter ICSP programming mode..." indefinitely.

nickgammon, 2018-04-02:
I must admit I haven't actually programmed an ATmega328PB because I don't have one to hand. However after checking the datasheets there don't seem to be any significant differences between the ATmega328P and the ATmega328PB.

Do you have decoupling capacitors between VCC and Gnd, and AVCC and Gnd? Typically you use 0.1 µF (100 nF) as close to the chip legs as you can get them. Removing "divide by 8" will, of course, speed the processor up, and then an absence of decoupling capacitors would have a greater effect.

There was a recent [issue about this](https://github.com/nickgammon/arduino_sketches/issues/49#issuecomment-377600795) and that person reported that using the decoupling capacitors fixed the problem. Also that was with the ATmega328PB, so it appears there is no programming reason for it not to work.

ryanoc3ros, 2018-04-03:
Hi Nick,
I appreciate the quick response. I have 100 nF caps as you have described, and checked them to make sure they were soldered correctly. I didn't find any issues there.

I tried to run your Board Detector program again, and strangely, it found the chip this time. I then ran the Board Programmer program, and it seemed to run smoothly, however when trying to load a sketch onto the ATmega328PB, it didn't respond. I ran Board Detector again, and this time, it couldn't find the chip, just like yesterday. Board Detector finds any of my other boards (ATmega2560, ATmega328P, etc) without issue.

I have attached the relevant section of my schematic, perhaps there is something obvious that I am missing.

Thanks for the support!
[image]

ryanoc3ros, 2018-04-03:
This gets stranger. So I added a diode between /RESET and 5V and ran both of your programs. It managed to detect the chip and burn the bootloader to it. I then plugged in my FTDI module and uploaded Blink.ino to my board. It seemed to work fine. I then unplugged the board from everything and powered it off of an external power supply, and it no longer blinked. Switching back and forth between the FTDI chip and external power resulted in the same behavior; blinking when hooked up to the computer, no response when on external power. After a few cycles of this, it started behaving strangely when hooked up to the FTDI board. Sometimes it would blink a few times and stop, sometimes it wouldn't blink at all. I ran your Detector and Programmer sketches, and it could not detect the chip anymore, just like before. The outputs from your programs when it initially worked is included below.

Your suggestion about connecting the external clock. Do you mean run a wire from XTAL1 on the master arduino, over to XTAL1 on the slave board?

As a side note, I briefly soldered the more widespread ATmega328P chip on the board yesterday, and everything worked perfectly, so I am fairly confident that power to the board is done properly. 

Will this bootloader make the ATmega328PB behave like a ATmega328P? i.e. no second UART, SPI, and I2C interfaces? If so, it might be worth it to just stick with the ATmega328P, unless there is another significant advantage to using this chip and bootloader.

These are the outputs from both your Board Detector and Board Programmer programs.

Atmega chip detector.
Written by Nick Gammon.
Version 1.20
Compiled on Apr  2 2018 at 20:01:56 with Arduino IDE 10805.
Attempting to enter ICSP programming mode ...
Entered programming mode OK.
Signature = 0x1E 0x95 0x16 
Processor = ATmega328PB
Flash memory size = 32768 bytes.
LFuse = 0xFF 
HFuse = 0xDE 
EFuse = 0xF5 
Lock byte = 0xEF 
Clock calibration = 0x8F 
Bootloader in use: Yes
EEPROM preserved through erase: No
Watchdog timer always on: No
Bootloader is 512 bytes starting at 7E00

Bootloader:

7E00: 0x11 0x24 0x84 0xB7 0x14 0xBE 0x81 0xFF 0xF0 0xD0 0x85 0xE0 0x80 0x93 0x81 0x00 
7E10: 0x82 0xE0 0x80 0x93 0xC0 0x00 0x88 0xE1 0x80 0x93 0xC1 0x00 0x86 0xE0 0x80 0x93 
7E20: 0xC2 0x00 0x80 0xE1 0x80 0x93 0xC4 0x00 0x8E 0xE0 0xC9 0xD0 0x25 0x9A 0x86 0xE0 
7E30: 0x20 0xE3 0x3C 0xEF 0x91 0xE0 0x30 0x93 0x85 0x00 0x20 0x93 0x84 0x00 0x96 0xBB 
7E40: 0xB0 0x9B 0xFE 0xCF 0x1D 0x9A 0xA8 0x95 0x81 0x50 0xA9 0xF7 0xCC 0x24 0xDD 0x24 
7E50: 0x88 0x24 0x83 0x94 0xB5 0xE0 0xAB 0x2E 0xA1 0xE1 0x9A 0x2E 0xF3 0xE0 0xBF 0x2E 
7E60: 0xA2 0xD0 0x81 0x34 0x61 0xF4 0x9F 0xD0 0x08 0x2F 0xAF 0xD0 0x02 0x38 0x11 0xF0 
7E70: 0x01 0x38 0x11 0xF4 0x84 0xE0 0x01 0xC0 0x83 0xE0 0x8D 0xD0 0x89 0xC0 0x82 0x34 
7E80: 0x11 0xF4 0x84 0xE1 0x03 0xC0 0x85 0x34 0x19 0xF4 0x85 0xE0 0xA6 0xD0 0x80 0xC0 
7E90: 0x85 0x35 0x79 0xF4 0x88 0xD0 0xE8 0x2E 0xFF 0x24 0x85 0xD0 0x08 0x2F 0x10 0xE0 
7EA0: 0x10 0x2F 0x00 0x27 0x0E 0x29 0x1F 0x29 0x00 0x0F 0x11 0x1F 0x8E 0xD0 0x68 0x01 
7EB0: 0x6F 0xC0 0x86 0x35 0x21 0xF4 0x84 0xE0 0x90 0xD0 0x80 0xE0 0xDE 0xCF 0x84 0x36 
7EC0: 0x09 0xF0 0x40 0xC0 0x70 0xD0 0x6F 0xD0 0x08 0x2F 0x6D 0xD0 0x80 0xE0 0xC8 0x16 
7ED0: 0x80 0xE7 0xD8 0x06 0x18 0xF4 0xF6 0x01 0xB7 0xBE 0xE8 0x95 0xC0 0xE0 0xD1 0xE0 
7EE0: 0x62 0xD0 0x89 0x93 0x0C 0x17 0xE1 0xF7 0xF0 0xE0 0xCF 0x16 0xF0 0xE7 0xDF 0x06 
7EF0: 0x18 0xF0 0xF6 0x01 0xB7 0xBE 0xE8 0x95 0x68 0xD0 0x07 0xB6 0x00 0xFC 0xFD 0xCF 
7F00: 0xA6 0x01 0xA0 0xE0 0xB1 0xE0 0x2C 0x91 0x30 0xE0 0x11 0x96 0x8C 0x91 0x11 0x97 
7F10: 0x90 0xE0 0x98 0x2F 0x88 0x27 0x82 0x2B 0x93 0x2B 0x12 0x96 0xFA 0x01 0x0C 0x01 
7F20: 0x87 0xBE 0xE8 0x95 0x11 0x24 0x4E 0x5F 0x5F 0x4F 0xF1 0xE0 0xA0 0x38 0xBF 0x07 
7F30: 0x51 0xF7 0xF6 0x01 0xA7 0xBE 0xE8 0x95 0x07 0xB6 0x00 0xFC 0xFD 0xCF 0x97 0xBE 
7F40: 0xE8 0x95 0x26 0xC0 0x84 0x37 0xB1 0xF4 0x2E 0xD0 0x2D 0xD0 0xF8 0x2E 0x2B 0xD0 
7F50: 0x3C 0xD0 0xF6 0x01 0xEF 0x2C 0x8F 0x01 0x0F 0x5F 0x1F 0x4F 0x84 0x91 0x1B 0xD0 
7F60: 0xEA 0x94 0xF8 0x01 0xC1 0xF7 0x08 0x94 0xC1 0x1C 0xD1 0x1C 0xFA 0x94 0xCF 0x0C 
7F70: 0xD1 0x1C 0x0E 0xC0 0x85 0x37 0x39 0xF4 0x28 0xD0 0x8E 0xE1 0x0C 0xD0 0x85 0xE9 
7F80: 0x0A 0xD0 0x8F 0xE0 0x7A 0xCF 0x81 0x35 0x11 0xF4 0x88 0xE0 0x18 0xD0 0x1D 0xD0 
7F90: 0x80 0xE1 0x01 0xD0 0x65 0xCF 0x98 0x2F 0x80 0x91 0xC0 0x00 0x85 0xFF 0xFC 0xCF 
7FA0: 0x90 0x93 0xC6 0x00 0x08 0x95 0x80 0x91 0xC0 0x00 0x87 0xFF 0xFC 0xCF 0x80 0x91 
7FB0: 0xC0 0x00 0x84 0xFD 0x01 0xC0 0xA8 0x95 0x80 0x91 0xC6 0x00 0x08 0x95 0xE0 0xE6 
7FC0: 0xF0 0xE0 0x98 0xE1 0x90 0x83 0x80 0x83 0x08 0x95 0xED 0xDF 0x80 0x32 0x19 0xF0 
7FD0: 0x88 0xE0 0xF5 0xDF 0xFF 0xCF 0x84 0xE1 0xDE 0xCF 0x1F 0x93 0x18 0x2F 0xE3 0xDF 
7FE0: 0x11 0x50 0xE9 0xF7 0xF2 0xDF 0x1F 0x91 0x08 0x95 0x80 0xE0 0xE8 0xDF 0xEE 0x27 
7FF0: 0xFF 0x27 0x09 0x94 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0x04 0x04 

MD5 sum of bootloader = 0xFB 0xF4 0x9B 0x7B 0x59 0x73 0x7F 0x65 0xE8 0xD0 0xF8 0xA5 0x08 0x12 0xE7 0x9F 
Bootloader name: optiboot_atmega328

First 256 bytes of program memory:

00: 0x0C 0x94 0x63 0x00 0x0C 0x94 0x8B 0x00 0x0C 0x94 0x8B 0x00 0x0C 0x94 0xBA 0x0F 
[... 28 lines cut ...]
HFuse = 0xDE 
EFuse = 0xF5 
Lock byte = 0xEF 
Clock calibration = 0x8F 
Bootloader address = 0x7E00
Bootloader length = 512 bytes.
Type 'Q' to quit, 'V' to verify, or 'G' to program the chip with the bootloader ...
Erasing chip ...
Writing bootloader ...
Committing page starting at 0x7E00
Committing page starting at 0x7E80
Committing page starting at 0x7F00
Committing page starting at 0x7F80
Written.
Verifying ...
No errors found.
Writing fuses ...
LFuse = 0xFF 
HFuse = 0xDE 
EFuse = 0xF5 
Lock byte = 0xEF 
Clock calibration = 0x8F 
Done.
Programming mode off.
Type 'C' when ready to continue with another chip ...

nickgammon, 2018-04-03:
> Your suggestion about connecting the external clock. Do you mean run a wire from XTAL1 on the master arduino, over to XTAL1 on the slave board?

No, I mean run a wire from D9 on the master Arduino (assuming it is a Uno) to XTAL1. The bootloader programmers uses a timer to output 8 MHz on that pin.

---

> Will this bootloader make the ATmega328PB behave like a ATmega328P? i.e. no second UART, SPI, and I2C interfaces?

No, not at all. All the bootloader does is let you upload code. Once the code is uploaded it transfers control to it, or if the bootloader times out (no incoming serial data) after about a second it transfers control to your code anyway.

The bootloader does not, in itself, add or subtract any functionality from your sketch. It's not like a "library" that provides services to running code. You can run code without the bootloader even being there.

---

> Switching back and forth between the FTDI chip and external power resulted in the same behavior; blinking when hooked up to the computer, no response when on external power. 

I would be checking your external power supply. Does it supply 5V exactly? What about ripple? Put your multimeter on AC and see what voltage it shows.

ryanoc3ros, 2018-04-03:
Without changing much, and burning the bootloader again, it seems to take sketches well now. Everything appears to be working. I am not quite sure why it is being inconsistent. Since this is a bug report for your programs, and that side of things seem to be working correctly, I will have to do more research and investigating to make it more stable. I appreciate all the help and advice you provided.

Thanks

nickgammon, 2018-04-03:
No problem. I use that code frequently to burn bootloaders so I don't think the code itself is at fault. I'll close this as probably a wiring issue.

closed 2018-04-03 by nickgammon