nickgammon/arduino_sketches #53: Atmega8 support

kapil-iitk, 2018-05-13:
Hi, I am trying to use ATMEGA_HEX_UPLOADER_FIXED_FILENAME with Atmega8 (with no crystal, Internal clock of 8MHz). I tried to modify your code, but all failed. Is it possible that you can share the code for the ATMEGA8 (with internal oscillator, 8MHz)? I am using the LED's and switch as discussed on your blog.

Thank you.

nickgammon, 2018-05-13:
Why did you modify the code? It supports the Atmega8. What does "all failed" mean? Did it not compile, or not run? If it didn't program the Atmega8 what results did you get? (Which LEDs flashed, etc.)

kapil-iitk, 2018-05-14:
Hi, Thanks for the reply, instead of ATMEGA328 I wanted to use ATMEGA8 as a brain of ARDUINO.  The compiled code occupy <14K which cannot use for the ATMEGA8. I have a lot of ATMEGA8 with me but not have ATMEGA328. Thanks

nickgammon, 2018-05-16:
Sure. If you look at the code where the words CROSSROADS_PROGRAMMING_BOARD occurs, there is provision for selecting something like 99 files with a rotary encoder and a LED readout.

kapil-iitk, 2020-05-12:
I designed a simplified schematic and PCB for the atmega328. Can you check the [schematic](https://github.com/kapil-iitk/AVR_SD_Programmer/blob/master/AVR_Prog.pdf)? Thank you.

nickgammon, 2023-02-21:
It is beyond the scope of offering this sketch to the public for me to check schematics, sorry.

closed 2023-02-21 by nickgammon