nickgammon/arduino_sketches #56: Atmega_Hex_Uploader_Fixed_Filename error codes

geologic, 2018-07-27:
Hi

I build a Atmega_Hex_Uploader_Fixed_Filename board.
Just to let you know if SD card board is not present (or not recognized), error led flashes 5 times, so you should update error code description to:

Red x 5 = Cannot read chip signature (Sg) OR SD card reader not present

Regards

nickgammon, 2018-07-27:
Actually it is one flash repeated 5 times. See http://www.gammon.com.au/forum/?id=11638&reply=5#reply5:

> **Flash rates**
>
> A "status" consists of one or more flashes, 200 mS apart (in other words, quickly blinking). The status groups are repeated with a one-second gap. So if you see 5 slow flashes (a second between each one) then that is "Cannot open SD card" (shown 5 times), not "Cannot reach chip signature" shown once. For example "Cannot enter programming mode" would be:
>
> `blink-blink-blink --- pause --- blink-blink-blink --- pause --- blink-blink-blink ... and so on, five times`

The idea of the repeated flashes is so that if you get (say) 4 flashes, you may be uncertain whether it was 3 or 4 flashes, so you get a second chance to count them (5 chances in fact).

So, one blink, with a 1s gap, and then another blink is the "number 1 error message".

geologic, 2018-07-30:
Ah... got it!
Case closed!

closed 2018-07-30 by geologic