nickgammon/arduino_sketches #58: bootloader erased when using hex_uploader

AdrienTouche, 2019-03-04:
Hello M. Gammond,
I'm self-learning arduino for years and putting now a foot in the bootload and fuse stuffs but I got a problem here.
I'm trying to use your hex_uploader sketch to upload on some home-made Arduino Pro Mini's from SD card. (ATMega328P, 5V, 16Mhz)
It succeds and the sketch uploaded works perfectly, but if I try to upload then something else from Arduino IDE the usual way it refuses.
Investigating with your wonderful Atmega_Board_Detector, I discovered that the Hex_Uploader is erasing my bootload (and then changing high fuse from 0xDA to 0xDB).
So using the Hex_Uploader with the version of the hex file of my sketch containing the bootloader, and then changing HFuse from 0xDB back to 0xDA I managed to make it work on the IDE uploads again.

But this bothers me. Should'nt the Hex_Uploader work properly with the standard hex file, without wiping out my bootload ?

Here is the Serial output of Hex_Uploader using the standard hex file : 
(notice that the output with the hex file with bootloader is quite the same
except 
#######################
Lowest address  = 0x0
Highest address = 0x7F9D
Bytes to write  = 14412
No bootloader.
Suggest making high fuse = 0xDB )

`
Atmega hex file uploader.
Written by Nick Gammon.
Version 1.37
Compiled on Feb 28 2019 at 15:51:41 with Arduino IDE 10805.
Reading SD card ...

HEX files in root directory:

MG_BLK0.HEX   :       1858 bytes.  Created: 2019-03-01 07:58:56.  Modified: 2019-03-01 07:55:22
MG_BLK1.HEX   :       4129 bytes.  Created: 2019-03-01 07:58:56.  Modified: 2019-03-01 07:49:54
MG_BLK2.HEX   :       4284 bytes.  Created: 2019-03-01 07:58:56.  Modified: 2019-03-01 07:54:08
MG_BLK3.HEX   :       4235 bytes.  Created: 2019-03-01 07:58:56.  Modified: 2019-03-01 07:54:34
MG_MSTR.HEX   :      68757 bytes.  Created: 2019-03-01 07:58:54.  Modified: 2019-03-01 07:56:14
PM_BLK0.HEX   :       1265 bytes.  Created: 2019-03-01 07:58:56.  Modified: 2019-02-28 14:09:28
PM_BLK1.HEX   :       2640 bytes.  Created: 2019-03-01 07:58:56.  Modified: 2019-02-28 14:10:58
PM_BLK2.HEX   :       2828 bytes.  Created: 2019-03-01 07:58:56.  Modified: 2019-02-28 14:13:04
PM_BLK3.HEX   :       2750 bytes.  Created: 2019-03-01 07:58:56.  Modified: 2019-02-28 14:13:44
PM_BLK4.HEX   :       2750 bytes.  Created: 2019-03-01 07:58:56.  Modified: 2019-02-28 14:14:16
PM_BT2-2.HEX  :      35498 bytes.  Created: 2019-03-01 08:59:06.  Modified: 2019-03-01 08:58:26
PM_BT2-3.HEX  :      35490 bytes.  Created: 2019-03-04 08:11:30.  Modified: 2019-03-04 08:10:16
PM_PB.HEX     :      30544 bytes.  Created: 2019-03-01 07:58:56.  Modified: 2019-03-01 07:47:10
BTBL_2-3.HEX  :      39692 bytes.  Created: 2019-03-04 10:25:50.  Modified: 2019-03-04 10:23:32

--------- Starting ---------

Attempting to enter ICSP programming mode ...
Entered programming mode OK.
Signature = 0x1E 0x95 0x0F 
Processor = ATmega328P
Flash memory size = 32768 bytes.
LFuse = 0xFF 
HFuse = 0xDA 
EFuse = 0xFD 
Lock byte = 0xCF 
Clock calibration = 0x9A 
Actions:
 [E] erase flash
 [F] modify fuses
 [L] list directory
 [R] read from flash (save to disk)
 [V] verify flash (compare to disk)
 [W] write to flash (read from disk)
Enter action:
Programming mode off.
W
Attempting to enter ICSP programming mode ...
Entered programming mode OK.

Choose disk file [ BTBL_2-3.HEX ] ...
PM_BT2-3.HEX
Processing file: PM_BT2-3.HEX
Checking file ...

####################
Lowest address  = 0x0
Highest address = 0x3141
Bytes to write  = 12610
No bootloader.
Suggest making high fuse = 0xDB 
Attempting to enter ICSP programming mode ...
Entered programming mode OK.
Processing file: PM_BT2-3.HEX
Erasing chip ...
Writing flash ...

################################################################
###################################
Written.
Processing file: PM_BT2-3.HEX
Verifying flash ...

################################################################
##################################
No errors found.
No bootloader.
Setting high fuse = 0xDB 
Done.

--------- Starting ---------

Attempting to enter ICSP programming mode ...
Entered programming mode OK.
Signature = 0x1E 0x95 0x0F 
Processor = ATmega328P
Flash memory size = 32768 bytes.
LFuse = 0xFF 
HFuse = 0xDB 
EFuse = 0xFD 
Lock byte = 0xFF 
Clock calibration = 0x9A 
Actions:
 [E] erase flash
 [F] modify fuses
 [L] list directory
 [R] read from flash (save to disk)
 [V] verify flash (compare to disk)
 [W] write to flash (read from disk)
Enter action:
Programming mode off.
`

nickgammon, 2019-03-05:
> But this bothers me. Should'nt the Hex_Uploader work properly with the standard hex file, without wiping out my bootload ?

It can't. The flash memory has to be erased before reprogramming and the erase instruction erases everything. There is no "erase one page" instruction. This can be done by the bootloader which does things in a different way.

AdrienTouche, 2019-03-05:
OK thanks for your reply. I hadn't understood this in the instructions and wikis.
But then is it normal that when I upload (with hex_uploader) the .hex file with bootload included,  it continues to say "No bootload" and changes the HFuse to disable the boot reset vector ?

I mean, if their is already a bootloader on the arduino it's erased. If their is one inside the .hex file it's ignored (but flashed nonetheless on the card) . So when does the hex_uploader recognize a bootlader  ?
Is it not something wrong with this when the .hex file contains the bootloader  ? : 
`
#######################
Lowest address = 0x0
Highest address = 0x7F9D
Bytes to write = 14412
No bootloader.
Suggest making high fuse = 0xDB 
`

the lowest adress or something ? Thks

nickgammon, 2019-03-11:
When the uploader finds code that starts at 0x0000 (as in your case) it assumes this code is not a bootloader, because bootloaders start at a high address. The issue is the fuses. You need to change the appropriate fuse bit to tell it to boot into the bootloader.

DirtyEngineer, 2019-03-11:
Also ensure you are uploading a hex file that includes the bootloader. If you are using the Arduino IDE, "Export compiled Binary" will generate two hex files, one with a bootloader and one without (labeled as such). Make sure you are using the correct one.

[image]

AdrienTouche, 2019-03-12:
> Also ensure you are uploading a hex file that includes the bootloader. 

Yes that's what I did. I was just surprised that the Hex_uploader wouldn't recognize it as a .hex WITH bootloader.
What I didn't know is that it seems that unlogicaly the bootloader is writen at the end of the memory, after the program itself. And the hex_uploader doesn't really search for e bootloader, it only looks what is the entry adress of the Hex file. As the first thing writen is the program (and not the bootloader) hex_uploader doesn't detect the bootloader writen at the end.
I suppose it would detect a bootloader only if it was writen alone, without any sketch attached.
In this case I know what I have to do : flash the .hex with booloader and then change the high fuse to ask the boot on the bootloader (or more simply block the automatic change of fuse in the hex_uploader)

Do I get it right ? If it is so I will declare it solved.

closed 2019-04-02 by AdrienTouche

nickgammon, 2019-06-14:
Yes, you could put both files into the one file for uploading and then change the fuse. It would be slower because you would also be uploading the gap between the sketch and the bootloader.

antno1000, 2019-12-27:
@AdrienTouche Hi, 

I too am facing the same problem, writing hex which includes bootloader makes Uno unable to be programmed via USB. Can you please help me figure out what can be done to resolve this problem? I am a beginner as well.
thanks

DirtyEngineer, 2019-12-27:
@antno1000 If you programmer a hex file with included bootloader you also need to program the fuse bit to allow it to run. What are your current fuse setting from your Uno?

antno1000, 2019-12-28:
@DirtyEngineer After i restored the bootloader the fuse bits restored to Lfuse: 0xFF; Hfuse: 0xDE; Efuse: 0xFD

When I programmed the hex file which included the bootloader fuses changed to Lfuse: 0xFF; Hfuse: 0xDF; Efuse: 0xFD
 and then I was no longer able to upload code via USB, unless i restored the bootloader once again.

DirtyEngineer, 2019-12-28:
@antno1000 You need to upload the hex file which includes the bootloader and then modify the Hfuse to 0xDE so it will run. You can do with with the HEX UPLOADER sketch ( "F" argument ) or use the command line to invoke avrdude to change them via ISP programmer. Even easier is the use Avrdudess which is a great GUI for avrdude. It can be found here.

antno1000, 2019-12-28:
@DirtyEngineer ok, can you guide me through on how can i do it automatically, maybe setting it inside programming arduino (maybe inside original source code) to rewrite fuse after it has flashed the other arduino?

DirtyEngineer, 2019-12-28:
@antno1000 Nick has done an excellent job of documenting the code so you shouldn't have much issue figuring out how add a highfuse write after programing the hex file.

antno1000, 2019-12-28:
@DirtyEngineer  Alright! thanks for helping out

AdrienTouche, 2019-12-29:
Hello ! @antno1000 I think that in the first page of the hex uploader script you can find something like "#nofusechange", just uncomment it should do the work if your atmega is already in an arduino configuration before.

antno1000, 2019-12-29:
@AdrienTouche Alright!, I'll give it a shot. BTW do you know how to do that with Fixed filename code?

AdrienTouche, 2019-12-31:
Same thing, the basis of the code is the same.

noeldum, 2024-07-15:
I came across the same issue and took me a while to understand the logic here. Everyone input was helpful to put me on track.

I now understand these sketches here do not use the mechanism offered by the Arduino IDE to make use of the concatenated hex file with bootloader.

Even if we provide a _with_bootloader_ hex file on the sd card it gets treated like a _without bootloader_ file. Hence disabling bootloader in the high fuse. 

If I am understanding right these sketches expect either a hex file or a separate bootloader file. Not both of them together in the same file. The difference is made using the _lowestAddress_ variable to detect which one is which. 

So any file (with or without bootloader) will see its fuse set to disable bootloader by the _updateFuses_ function. 
Clearly feels counter intuitive at first. Hence why some of us wanting to automate this and have it enabled at flashing time.

Is there really a need to set the fuses once flashed? From the code _updateFuses_ only deals with the bootloader part.

I worked around the problem by commenting out _updateFuses_ and add my own fuse setup after that.
writeFuse(0xDE, highFuse);

Ideally the sketch could maybe detect a _with_bootloader_ file and decide to enable bootloader in this particular case. A suggestion to consider maybe.