nickgammon/arduino_sketches #60: ATMega2560 strange fuse issue when uploading hex with bootloaders

AdrienTouche, 2019-04-04:
Hello.
This question seems related to the topic  of travisfarmer about "HEX uploader on CrossRoads programmer: verification error when programming Mega2560 with bootloader", but my setup is quite different.
I'm using the Hex_uploader, without CrossRoads platform, to make updates on a system of multiple arduinos. 5 Atmega328P are talking to a Mega2560 through I2C, and all of them are linked by Miso Mosi and Sck to a 6th 328P with a slightly modified Hex-uploader. The modifications are of course to be able to change the RESET pin in the software to be able to choose on which arduino to make each update.
This works quite fine with all the ATMega328P, but the Mega2560 has a very strange behaviour.
I made 2 systems of this type, identical in their wirings (1 for early tests with arduinos linked with jumpers on a breadboard, and the second with a dedicated soldered PCB), and the 2 give me each a different behaviour of the Mega2560. On the first one like in the travisfarmer bug it almost always fails on the verification (of a hex file WITH bootloader) and the sketch don't work. BUT I recently discovered that it stops failing if the HIGH fuse is already at 0xD1 (and not 0xD0 like it is for the booting on bootloader). If I change the fuse before uploading, it succeeds and the sketch works ! And then I put back my Hfuse to 0xD0.
The second system fails everytime (probably due to bad connections in my solderings) but in addition (more strangely) trying to upload a sketch sometimes changed randomly the fuses ! (even after modifying the Hex-Uploader code to forbid it to change fuses)
The result of course is that it ended bricking my Mega2560...
All this to say that it seems that in the case of the Mega2560 the fuses and the sketch/bootload flash space are related... and it's no good. What do you think ?

AdrienTouche, 2019-04-04:
I forgot to say that I tried of course my installations with the unmodified Hex_Uploader sketch (uploading only to the Mega2560) to see if the behaviour would be the same and it was. So the trouble doesn't seem to come from my modifications.
And concerning the second system, I think the bad connections altered the SPI communications, resulting with bad readings from the mega2560. But it still seems odd that it ends modifying the fuses during an upload...

AdrienTouche, 2019-04-23:
Hi ! So would anyone have an idea about this ? It's really inconvenient for what I try to do.

nickgammon, 2019-06-12:
Sorry about the late response, I was on holiday about the time you posted this, and got a bit distracted. It sounds like a hardware issue to me, possibly not enough decoupling capacitors on Vcc etc.

AdrienTouche, 2020-01-02:
Hi !  I try to get a fresh view on this topic because I got stuck on this problem and I'm short of good ideas to solve it. 
It's been a long time since you answered me to check the decoupling capacitors, I was kind of unable to check this possibility until I recently bought a Picoscope Oscilloscope with data signals.
It shows that surely my home made Arduino 2560 are not that strong in maintaining a flat and good Vcc signal but this doesn't seem to be the only problem, since using an official Arduino mega board doesn't work well either.
I think now that there is 2 different problems in 1.

1. A bug of the Hex_uploader : it can't write or verify correctly a hex file with bootloader on a ATMega2560 if the high fuse is on D8 or D0 (classic Arduino configuration for instance). BUT it works perfectly well if you change before doing anything this High Fuse to D1 or D9... and then put it back. (for information this is enabling/disabling the bootloader)

(Note :  this can be wrongly analysed as a wiring issue, because if you try to write with the High Fuse  on D8 for instance, it won't work. BUT as the program can't see the bootloader after the sketch on the hex file it will assume there is noone, and will automatically change the High Fuse to D9. SO when you try it the second time it works !!! You see it as a wiring issue but in fact it's the change of fuse which made it work.
I think that's exactly what happened to Mr Gammon on this post : 
https://github.com/nickgammon/arduino_sketches/issues/47#issuecomment-340998761
)

2. Even knowing this, I continue to have troubles on my particular setup. The idea of it is to have 4 Atmega328P and 1 AtMega2560 all sharing Miso Mosi and SCK with the Bit Bang SPI of a Master 328P (with the Hex_uploader). When I want to upload a hex file on one of them I use it's Reset pin as the SS line and keep the other Resets high.
It works perfectly fine for the AtMega328P's but keeps being very instable with the AtMega2560 (worst case scenario being the bricking of the 2560 during verifying...). Readings with the picoscope seem to indicate a failure of the response of the 2560 on the MISO line .
My question is could the Miso pins of the 328P's being on high impedance pollute the signal of the 2560? Would it be useful for instance to add a pullUp resistance on Miso and Mosi ?

AdrienTouche, 2020-05-02:
Hello ! I'm trying to wake this thread up a bit... Any ideas to make the AtMega2560 hex upload more stable ? Sharing the SPI with other chips should not get any problems right ? (as long as they all have their separate SS line). Could the Bit Banged SPI generation pollute the quality of the signal in this case? or use another chip than  a 328P as master could help ?

AdrienTouche, 2020-05-02:
I must add that adding decoupling capacitors doesn't seem to help (thought it's really hard to be sure)

nickgammon, 2020-05-03:
So, to be clear, you are uploading a really large file, which consists in part of a sketch (code) to be executed, as well as a  bootloader? The uploader was originally designed to be used in place of the bootloader, or to install a bootloader. It wasn't designed to upload both code and a bootloader.

nickgammon, 2020-05-03:
And the failure is in the process of burning this large file? Or it seems to burn it (no error messages) but the bootloader subsequently does not work with the Arduino IDE? Is that right?

AdrienTouche, 2020-05-03:
Hi !Thank you so much to take time on this M. Gammon. I didn't know that the uploader wasn't originally designed to upload both code and a bootloader. My idea was to use it to simplify the uploading process (just put a SD card in the slot and the update is automatic) but to keep the arduino bootloader in order to let the avrdude access available too. Maybe it's too much. I need to try an upload of a big sketch without bootloader to see if it works better. I will do that tomorrow.

AdrienTouche, 2020-05-03:
Concerning the failure yes it's during the uploading process, and it says at the verifying step that its wrong.  Actually my last bricked Mega2560 chip happened during a **verify only** of the hex_uploader. The sketch (with the bootloader) was already burned (succesfully) on the chip. So if it's not a decoupling capacitor problem it could be pollution from the other slaves on the SPI lines. Maybe I will try [these improvements](https://www.pjrc.com/better-spi-bus-design-in-3-steps/) to see what happens...

AdrienTouche, 2020-05-04:
Hello ! It has been driving me crazy but I think I understood some things.
First I tried again with High Fuse at D0 or D8 (means bootloader enabled). It systematicaly fails :

> --------- Starting ---------
> 
> Attempting to enter ICSP programming mode ...
> Entered programming mode OK.
> Signature = 0x1E 0x98 0x01 
> Processor = ATmega2560
> Flash memory size = 262144 bytes.
> LFuse = 0xFF 
> HFuse = 0xD0 
> EFuse = 0xFD 
> Lock byte = 0xFF 
> Clock calibration = 0xA0 
> Actions:
>  [E] erase flash
>  [F] modify fuses
>  [L] list directory
>  [R] read from flash (save to disk)
>  [V] verify flash (compare to disk)
>  [W] write to flash (read from disk)
> Enter action:
> Programming mode off.
> W
> Attempting to enter ICSP programming mode ...
> Entered programming mode OK.
> 
> Choose disk file [ BTESTB.HEX ] ...
> BVOIDB.HEX

> 
> Processing file: BVOIDB.HEX
> Checking file ...
> 
> #############
> Lowest address  = 0x0
> Highest address = 0x3FD1D
> Bytes to write  = 8110
> No bootloader.
> Suggest making high fuse = 0xD1 
> Attempting to enter ICSP programming mode ...
> Entered programming mode OK.
> Processing file: BVOIDB.HEX
> Erasing chip ...
> Writing flash ...
> 
> #################################
> Written.
> Processing file: BVOIDB.HEX
> Verifying flash ...
> Verification error at address 0. Got: 0xFF  Expected: 0x71 
> Verification error at address 1. Got: 0xFF  Expected: 0xC0 
> Verification error at address 2. Got: 0xFF  Expected: 0x00 
> Verification error at address 3. Got: 0xFF  Expected: 0x00 
> Verification error at address 4. Got: 0xFF  Expected: 0x81 
> Verification error at address 5. Got: 0xFF  Expected: 0xC0 
> Verification error at address 6. Got: 0xFF  Expected: 0x00 
> Verification error at address 7. Got: 0xFF  Expected: 0x00 
> Verification error at address 8. Got: 0xFF  Expected: 0x7F 
> Verification error at address 9. Got: 0xFF  Expected: 0xC0 
> Verification error at address A. Got: 0xFF  Expected: 0x00 
> Verification error at address B. Got: 0xFF  Expected: 0x00 
> Verification error at address C. Got: 0xFF  Expected: 0x7D 
> Verification error at address D. Got: 0xFF  Expected: 0xC0 
> Verification error at address E. Got: 0xFF  Expected: 0x00 
> Verification error at address F. Got: 0xFF  Expected: 0x00 
> Verification error at address 10. Got: 0xFF  Expected: 0x7B 
> Verification error at address 11. Got: 0xFF  Expected: 0xC0 
> Verification error at address 12. Got: 0xFF  Expected: 0x00 
[... 88 lines cut ...]
> Done.
> 
> --------- Starting ---------
> 
> Attempting to enter ICSP programming mode ...
> Entered programming mode OK.
> Signature = 0x1E 0x98 0x01 
> Processor = ATmega2560
> Flash memory size = 262144 bytes.
> LFuse = 0xFF 
> HFuse = 0xD1 
> EFuse = 0xFD 
> Lock byte = 0xFF 
> Clock calibration = 0xA0 
> Actions:
>  [E] erase flash
>  [F] modify fuses
>  [L] list directory
>  [R] read from flash (save to disk)
>  [V] verify flash (compare to disk)
>  [W] write to flash (read from disk)
> Enter action:
> Programming mode off.

But as you see the program automaticaly change the fuse to D1 so next time it works.

AdrienTouche, 2020-05-04:
Then, with High Fuse at D1 it worked quite well, with one of my home made Mega2560 arduino.
So I tried an other one home made Mega2560 with a different design (integrating some I2C chips in addition for Led driving functions. It failed first, then second time, and bricked... Chip is not responding to any ISP or Serial command...
So I tried again the first chip, and know it didn't work either. So it seems to be wiring issues in the Vcc and Gnd lines. But what is strange is that during failing process it also changes some fuses randomly, until it finishes bricking it.
I attach down here the Serial output of the last trials, until it bricked too. You can see the fuses changing on their own will, and the output of the verification (always the same : 00 00 01 01 02 02 03 03, etc...)

> 
> --------- Starting ---------
> 
> Attempting to enter ICSP programming mode ...
> Entered programming mode OK.
> Signature = 0x1E 0x98 0x01 
> Processor = ATmega2560
> Flash memory size = 262144 bytes.
> LFuse = 0x8F 
> HFuse = 0xD1 
> EFuse = 0xFD 
> Lock byte = 0xFF 
> Clock calibration = 0xA0 
> Actions:
>  [E] erase flash
>  [F] modify fuses
>  [L] list directory
>  [R] read from flash (save to disk)
>  [V] verify flash (compare to disk)
>  [W] write to flash (read from disk)
> Enter action:
> Programming mode off.
> W
> Attempting to enter ICSP programming mode ...
> Entered programming mode OK.
> 
> Choose disk file [ BVOIDB.HEX ] ...
> BTEST.HEX
> Processing file: BTEST.HEX
> Checking file ...
> 
> ###########################################
> Lowest address  = 0x0
> Highest address = 0x69AD
> Bytes to write  = 27054
> No bootloader.
> Suggest making high fuse = 0xD1 
> Attempting to enter ICSP programming mode ...
> Entered programming mode OK.
> Processing file: BTEST.HEX
> Erasing chip ...
> Writing flash ...
> 
> ################################################################
> ##########################################
> Written.
> Processing file: BTEST.HEX
> Verifying flash ...
> Verification error at address 0. Got: 0x00  Expected: 0x44 
> Verification error at address 1. Got: 0x00  Expected: 0xC1 
> Verification error at address 2. Got: 0x01  Expected: 0x00 
> Verification error at address 3. Got: 0x01  Expected: 0x00 
> Verification error at address 4. Got: 0x02  Expected: 0x73 
> Verification error at address 5. Got: 0x02  Expected: 0xC1 
> Verification error at address 6. Got: 0x03  Expected: 0x00 
> Verification error at address 7. Got: 0x03  Expected: 0x00 
> Verification error at address 8. Got: 0x04  Expected: 0x71 
> Verification error at address 9. Got: 0x04  Expected: 0xC1 
> Verification error at address A. Got: 0x05  Expected: 0x00 
> Verification error at address B. Got: 0x05  Expected: 0x00 
> Verification error at address C. Got: 0x06  Expected: 0x6F 
> Verification error at address D. Got: 0x06  Expected: 0xC1 
> Verification error at address E. Got: 0x07  Expected: 0x00 
> Verification error at address F. Got: 0x07  Expected: 0x00 
> Verification error at address 10. Got: 0x08  Expected: 0x6D 
[... 234 lines cut ...]
> Verification error at address 5E. Got: 0x2F  Expected: 0xCB 
> Verification error at address 5F. Got: 0x2F  Expected: 0x0F 
> Verification error at address 60. Got: 0x30  Expected: 0x45 
> Verification error at address 61. Got: 0x30  Expected: 0xC1 
> Verification error at address 62. Got: 0x31  Expected: 0x00 
> Verification error at address 63. Got: 0x31  Expected: 0x00 
> Verification error at address 64. Got: 0x32  Expected: 0x0C 
> 
> ################################################################
> #########################################
> 26935 verification error(s).
> First 100 shown.
> No bootloader.
> Setting high fuse = 0x01 
> Done.
> 
> --------- Starting ---------
> 
> Attempting to enter ICSP programming mode ......................................................
> Failed to enter programming mode. Double-check wiring!
> Halted.
> Programming mode off.
> 
> 
>

AdrienTouche, 2020-05-04:
So what do you think ? I think I will try to heavily add decoupling capacitors, and to add a High Voltage Parallel Programming connection on a new design, because it's beginning to cost me a lot of ATMega2560 chips... 

Do you know if their is a maximum number of decoupling capacitors we can add to a circuit ? And what values are best ? (I have always put only 100nF, but perhaps I shoud change some ?)

nickgammon, 2020-05-16:
I have found that when I get these issues it is hardware. I know you think your hardware is OK, but the uploader worked for me many times with the Atmega328P, and also the Atmega2560. In fact, on one occasion I had problems with a client who had a custom board made with a Atmega2560. The problem was with the design of the reset circuitry (which took days to find) and *not* the uploading software.

> Do you know if there is a maximum number of decoupling capacitors we can add to a circuit ? 

I don't think there is any limit except the practicalities of having capacitors everywhere. 100 nF is the recommended value, and the capacitor type should be ceramic, although I personally have used other types without problems.

There should be a short run between the capacitor and Vcc and Gnd. Too large a run will introduce problems.

If possible, slow down the clock rate for the uploading. If you are using the SPI uploading try changing the speed:

```c
SPI.setClockDivider (SPI_CLOCK_DIV64);  // <-- something slower
```

closed 2023-02-21 by nickgammon