nickgammon/arduino_sketches #61: Atmega Hex Uploader through uart

isnadh, 2019-07-12:
Hello Nick! I have been following your works for quite some time and I must confess they are top notch. **I recently used your BigNumber library for my thesis**!.

Could you kindly point me in the right direction on how to modify your ATMEGA hex uploader to allow burning through UART (loaded with optiboot) instead of SPI.

I will be so delighted if you can help out on this feat.

nickgammon, 2019-07-16:
It wouldn't be a modification, it would be a rewrite. Using SPI is totally different than using the UART. One of the reasons I wrote the loader was to upload sketches to chips which didn't (or couldn't) have a bootloader.

isnadh, 2019-07-21:
I see. Thanks a lot.

closed 2019-08-13 by isnadh