nickgammon/arduino_sketches #63: Doesn't work with Atmega328PB

crobertsbmw, 2019-09-09:
Thanks for your support and putting this together.

I got things working with a 328P on a breadboard. But when I swapped to the 328PB things don't seem to work. I was able to use the Board_Detector and correctly identify the board one time. Then I swapped to the Board_Programmer, but I just got the timeout every time.  

I then went back to the Board_Detector and that stopped working and gave me the timeout every time as well. Maybe I screwed something up in between?

nickgammon, 2019-10-01:
Usually this is caused by hardware issues, for example:

* None, or indadequate, decoupling capacitors on Vcc
* Incorrect fuse settings on the processor
* Noise, possibly introduced by the breadboard
* Too long leads (eg. between the crystal/resonator) and the processor
* Incorrect power supply voltage, or noisy power supply
* Missing clock - for example if you have the fuses set to expect a crystal but don't have one on the board

crobertsbmw, 2019-10-02:
Thanks for your response. I can't quite remember, but I think I was forgetting the decoupling capacitor and that got it working. Thanks again for this great library.

closed 2019-10-02 by crobertsbmw