nickgammon/arduino_sketches #64: Atmega_Board_Programmer:143:10: fatal error: HV_Pins.h: No such file or directory Error

cglkr97, 2020-03-23:
Hey guys!

I followed this tutoria on Youtube and checked my wirings 10 times but still got this compiler error.  
This is the tutorial:
https://www.youtube.com/watch?v=dpgcBsl9D4k

And this is the complete error:

**_Atmega_Board_Programmer:143:10: fatal error: HV_Pins.h: No such file or directory

 #include "HV_Pins.h"

          ^~~~~~~~~~~

compilation terminated.

exit status 1
HV_Pins.h: No such file or directory_**

What should i do about this?

nickgammon, 2020-03-23:
That file is in the repository:

https://github.com/nickgammon/arduino_sketches/blob/master/Atmega_Board_Programmer/HV_Pins.h

You must have not downloaded all the files correctly.

cglkr97, 2020-03-24:
Yes, I am aware you have that file in the repository. Forgive me if I sound awfully ignorant, I'm a college student. But is this supposed to be a library? My guess was it is not a library since it has header files but no .cpp files. It gave an error when I tried to add it as a .zip library anyways.  But if it isn't a library then how the IDE suppose to know and locate the HV_Pins.h?

Am I missing steps here? Is there anything else that I didn't do before uploading the code software-wise. Or is there anything else that you think might have caused this problem. Sorry if that sounded so dumb.

nickgammon, 2020-03-24:
It is not a library. In this folder: https://github.com/nickgammon/arduino_sketches/tree/master/Atmega_Board_Programmer all the files are in the same folder. The compiler simply includes them, as normal, because they are in the same directory. You should have all those files in one folder, and then open Atmega_Board_Programmer.ino in the IDE, and compile it. Everything should follow from that. The IDE concatenates all the .ino files into a .cpp file and compiles that.

I have a description of that process here: https://arduino.stackexchange.com/a/13182/10794

cglkr97, 2020-03-25:
I solved the problem finally! I simply extracted the zip file to a normal unzipped file and then opened the code. I know how dumb that sounds :|  I successfully uploaded your bootloader sketch to my board and it worked smoothly. I have been struggling with this bootloading problem for days and it was driving me crazy! I can not thank you enough for putting together this code! You are a hero!!

nickgammon, 2020-03-25:
I'm glad you got it to work. Just to summarise for anyone else reading this, you download the .zip file, extract the whole thing into a folder, and then open the file Atmega_Board_Programmer.ino in the IDE and compile it.

closed 2020-03-25 by nickgammon