nickgammon/arduino_sketches #65: Bootloading a 8MHz(External clock) 3.3V ATmega328P

cglkr97, 2020-03-26:
Hello! 

I successfully bootloaded my Atmega328P thanks to Nick's helps. Bootloading went smootly and I bootloaded my chip using the Lilypad 8MHz configuration as I wanted to operate my chip using 8 MHz external clock and 3.3V on a breadboard.  After succesfully bootloading it, I conencted my bootloaded breadboard Atmega328P to an Ardunino UNO board (chip removed) to program it and I always got this error:

**_avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x33_**

Later on, when going over Atmega_Board_Programmer code, I realized that Lilypad 8MHz configuration means to use the internal clock. That's not suitable to me. So I rebootloaded my chip with UNO configuration and tried programming it using my UNO board again. Yet I still got the same error. Therefore I was able to bootload my Atmega328P but not been able to progam it.

Long story short, why do I see this error? How can I bootload my chip so that I can use it as 8MHz 3.3V chip?

PS: My programming connections to UNO boards are like this:

Pin 0 on UNO (Rx) ----> Pin 2 on Atmega328P
Pin 1 on UNO  (Tx) ----> Pin 3
Reset on UNO ----> Pin 1 (Also a 10k Pull-up resistor here)
8MHz crystal on Atmega328P Pin 9 & Pin 10 on the other leg connected 22 pF caps that go to ground.
Pin 7 and Pin 20  going to Vcc
Pin 8 & Pin 22 going to ground
And I have a working MCP1702 3.3V voltage regulator that succesfully supplies 3.3V to my board.

nickgammon, 2020-03-26:
When you attempted to upload your sketch in the IDE, what type of board did you tell it you had? That affects the speed at which the commands are sent to the chip.

> 8MHz crystal on Atmega328P Pin 9 & Pin 10 on the other leg connected 22 pF caps that go to ground

How are you doing that? The Uno board has a resonator on it.

> How can I bootload my chip so that I can use it as 8MHz 3.3V chip?

You want the crystal, is that it? For accuracy? You can upload the Lilypad bootloader and tweak the fuse to make it use the crystal. Or upload the Uno bootloader, and change the baud rate in the boards.txt file so it sends the commands more slowly.

Your problem is basically a baud rate mismatch between what the bootloader expects and what your programming board is sending.

cglkr97, 2020-03-26:
`When you attempted to upload your sketch in the IDE, what type of board did you tell it you had? That affects the speed at which the commands are sent to the chip.`

I first selected the Arduino Pro Mini 8MHz 3.3V board assuming I burned my bootloader like that (or baudrate in this case then) . It obviously didn't work. So I went on trying MiniCore. From the MiniCore boards, I selected Atmega328P with 8 MHz external oscillator choice. The following were my selections in MiniCore settings:

Board: Atmega328
Clock: External 8MHz
BOD: BOD 1.8V
Compiler LTO: LTO disabled
Variant: 328P/328PA
Bootloade: Yes(UART0)

Yet this also did not work. 

`How are you doing that? The Uno board has a resonator on it.`

Yes, you are right. It does have an 16MHz clock on it. But I still placed a 8MHz clock on my breadboard Atmega328P and connected an UNO to program it.  Do you think that might also interfere with the programming? 
I actually didn't want to use my UNO board as my programmer. At the beginning of this problem when my FTDI chip didn't upload code, I though that this was a mere FTDI failure so I switched to UNO board as my programmer. But now we know that both FTDI chip and the UNO board fails to
upload code. So your guess must be correct, baudrate difference should be the issue here.  

`You can upload the Lilypad bootloader and tweak the fuse to make it use the crystal.`

So I need to change the fuses to whatever a chip with external 8MHz clock would have as fuse setting. Like google the fuse settings for Arduino Pro Mini and change the fuses accordingly. Would that work? I know there is a fuse calculator but I couldn't wrap my mind around it. 

`Or upload the Uno bootloader, and change the baud rate in the boards.txt file so it sends the commands more slowly.`

You mean the upload.speed setting in boards.txt? So I need to go there and change the UNO's baudrate to something lower. Someting lower like what? Whavever Pro Mini uses for example?

Sorry to bother you with many questions. But to study this stuff is somehow the only thing that makes me feel better in these locked up days. Thanks!

nickgammon, 2020-03-27:
> When you attempted to upload your sketch in the IDE, what type of board did you tell it you had? That affects the speed at which the commands are sent to the chip.
>
> I first selected the Arduino Pro Mini 8MHz 3.3V board assuming I burned my bootloader like that (or baudrate in this case then) 

In the Arduino IDE I would select Uno if you did the Uno bootloader, and Lilypad if you did the Lilypad bootloader. You don't need to overthink that part.

> But I still placed a 8MHz clock on my breadboard Atmega328P and connected an UNO to program it. Do you think that might also interfere with the programming?

It doesn't really matter what you are programming it with. If you are using a breadboard, you need a 16 MHz crystal if you selected the Uno bootloader, and no crystal (although having one won't hurt) if you select the Lilypad bootloader.

> You mean the upload.speed setting in boards.txt? So I need to go there and change the UNO's baudrate to something lower. Someting lower like what? Whavever Pro Mini uses for example?

First, I would copy/paste that whole Uno section and call it something else. For example, every line starting with uno.xxxx I would copy/paste and call it myuno.xxxx or similar.

Also change its name so you can tell one from the other in the IDE, eg.

```
myuno.name=My breadboard Uno
```

Then, since you are running at 8 MHz and not 16 MHz halve the baud rate. So, change:

```
myuno.upload.speed=115200
```

to be:

```
myuno.upload.speed=57600
```

Half the clock, half the baud rate.

cglkr97, 2020-03-28:
`First, I would copy/paste that whole Uno section and call it something else. For example, every line starting with uno.xxxx I would copy/paste and call it myuno.xxxx or similar.`

So I tried this and I still am not able to upload code to the chip. You can see my fuses and my boards.txt down below. 

[image]

I know the lock byte functions different than the fuses. But do you think mismatching lock bytes between the boards.txt and the bootloader might have anything to do with this problem? Because at this point (after trying many things) I can't think of anything else. 

And you can see my little programming board down below in case you would have questions about my wirings.

[image]

nickgammon, 2020-03-29:
* I would set myuno.build.f_cpu = 8000000, as that is the clock speed you are running at. However I think that would only affect the running sketch, not uploading it. 

* Your 0.1 µF capacitors are larger than I am accustomed to. 

* You seem to have one of the capacitors on the crystal between 5V and the crystal leg instead of Gnd. 

* I can't really see your wiring around the crystal part.

* There needs to be a 0.1 µF capacitor in series with RTS and the Reset pin (pin 1) which I can't see in your picture. 

* What are those parts in the corner? Is that a 3.3V voltage regulator? Try removing that for the moment and testing with straight 5V in case there is something wrong with that.

cglkr97, 2020-03-29:
Hello again Nick.

Here goes the changes I made so far. I set myuno.build.f_cpu = 8000000, removed the 0.1uF caps since I found out that they are not really a must.

`You seem to have one of the capacitors on the crystal between 5V and the crystal leg instead of Gnd.`

Yes, I realized that right after I posted the comment. Fixed it. 

`What are those parts in the corner? Is that a 3.3V voltage regulator? Try removing that for the moment and testing with straight 5V in case there is something wrong with that.`

Yes, it is a 3.3V voltage regulator. It succesfully provides 3.3V to the board. I measured it time and time again with my multimeter and it always shows around 3.3V. Yet, still I removed it and tried with  5V from the FTDI. 

Here's my board. Everything seem to match to your board except for the crystal and the caps between Vcc and ground. But I added the caps knowing it wouldn't make a big difference. 

So, long story short, after doing all these steps I still have a board that refuses to upload code. 

[image]

PS: Sorry for the bad angle of the picture. I take it with straight angle but when I post it, it looks a bit different somehow. It's like it gets rotated.

nickgammon, 2020-03-29:
I don't know. It ought to work. You should really have the 0.1 µF capacitors as shown in my photo. I just commented that yours looked rather large, and perhaps were too far away. They really should be on the legs of the appropriate pins.

See [What happens when code is uploaded using the bootloader?](https://arduino.stackexchange.com/questions/15936/what-happens-when-code-is-uploaded-using-the-bootloader).

If you are using the Uno bootloader you should see an LED on pin D13 (pin 19 on the chip) flash 3 times. You should use a resistor (eg. 470 ohms) in series with that LED. If not, then chip is not running for some reason.

You could also put LEDs (and resistors) on the serial pins (2 and 3) and see what you see. For example, you should see data arriving on Rx (pin 2) and some sort of response on Tx (pin 3).

If it was me I would put a scope or logic analyser on the serial pins (2 and 3) and see what the heck is happening, if anything.

tr102070, 2020-03-31:
Hi cglkr97,
are you sure you have a bootloader loaded on that chip? in order for FTDI to work your Atmega has to have a bootloader installed first.
good luck.

cglkr97, 2020-04-05:
Hey Nick!

Sorry, took me so long to get back to you. I solved the problem finally. The chips work perfectly. As you advised I added my Breadboard Arduino to the boards.txt by copying the UNO board and changing to baudrate to half. And then I went ahead uploaded your bootloader as Lilypad. And when I uploaded code to the chip (chose the Breadboard Arduino as my board of course) it was successful. So everything works great, thanks for all your help! But right now I'm also little confused even after solving the problem because after uploading the bootloader my fuses looked like this,

LFuse = 0xE2 
HFuse = 0xDA 
EFuse = 0xFD 
Lock byte = 0xEF 
Clock calibration = 0xA2 

Yet in the boards.txt my Breadboard Arduino has the following fuse settings since I copied and pasted it from the UNO part,

myuno.bootloader.low_fuses=0xFF
myuno.bootloader.high_fuses=0xDE
myuno.bootloader.extended_fuses=0xFD
myuno.bootloader.unlock_bits=0x3F
myuno.bootloader.lock_bits=0x0F

I'm asking just to get more insight into the Arduino chips. So the fuse settings of my board and my bootloader does not match. How come am I able to upload code to my chip if the fuses do not match?

nickgammon, 2020-04-05:
As I mentioned above in the link about how the bootloader works, the fuse settings in boards.txt are sent to the chip but ignored. So it doesn't matter what they are.

closed 2023-02-21 by nickgammon