nickgammon/arduino_sketches #69: Atmega_Board_Detector.ino not compiling

A-Broadbent, 2020-08-25:
Hi Nick, I downloaded the file & got such a large number of errors when trying to compile (stray characters of all types) that I gave up, have you a "clean" copy of the file, or am I somehow crooking it? I downloaded it directly & I'm using Arduino IDE 1.8.12

many tnx

nickgammon, 2020-08-25:
That should compile fine. What "file" are you referring to? You have to unzip the big file that gets downloaded. What operating system are you using?

If you look [here](https://github.com/nickgammon/arduino_sketches/tree/master/Atmega_Board_Detector) you will see 13 files. Apart from .gitignore they are all used in the compilation process. You should have those files on your disk, looking exactly the same as those 13 files (click on each link to see what they look like).

A-Broadbent, 2020-09-22:
Hi Nick, my job has been depriving me of the time needed to fiddle about with this...

I was downloading individual files, & getting tons of error messages such as:

[cid:3fdd65ca-a66e-42de-8222-06703efaa6ac]
however, after downloading the whole pack & unzipping, everything is compiling fine; a very useful set of tools, congrats & thanks!

Antonio R. Sánchez de Cañete Broadbent
https://es.linkedin.com/pub/antonio-rafael-s%C3%A1nchez-de-ca%C3%B1ete-broadbent/b/825/902

[https://ipmcdn.avast.com/images/icons/icon-envelope-tick-green-avg-v1.png]<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>     Libre de virus. www.avg.com<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>

closed 2023-02-21 by nickgammon