nickgammon/arduino_sketches #71: Trying to program multiple target arduino chip at once

akshulg, 2020-11-09:
I am trying to program multiple chips at once. Made a prototype by multiplexing chip select pin of the master arduino on which this program will be uploaded. So far its not working. I also tried connecting RESET pin of target arduino to its chip select pin but that is also not working. Any help?

nickgammon, 2020-11-10:
You won't be able to do that. Programming each chip requires a response from it. Tying all the MISO/MOSI/SCK lines together won't work. For one thing, MISO are all inputs to the programming board, so they will "fight" each other. The programming individually only takes a few seconds, so I wouldn't waste time trying to do them in bulk.

akshulg, 2020-11-10:
So what I am trying now is to connect RESET of other targets to gnd when programming the first one. This works but have to figure out how to automate this. Also getting some weird behaviour as well so trying to figure out if this will work. I am discussing this on this forum - https://arduino.stackexchange.com/questions/79199/programming-multiple-arduino-at-once
It will be helpful if you can give some feedback. Thanks

closed 2023-02-21 by nickgammon