nickgammon/arduino_sketches #72: GSM Based FOTA Block Diagram

akkoyun, 2020-12-23:
Hi all,
I want to build an AVR system that can be update firmware over a GSM-based IoT cloud. I draw a block diagram below. 

1. Main MCU sends a current firmware version to the cloud.
2. If new firmware released for that device cloud returns an ID for the new firmware file.
3. The Main MCU opens the SD communication line with SD_SEL pin HIGH.
4. The Main MCU downloads the new firmware to SD Card.
5. After download the main MCU closes the SD communication line with SD_SEL pin LOW.
6. the Main MCU sets the FOTA_EN pin HIGH for powering the FOTA MCU.
7. FOTA MCU gets powered.
8. FOTA MCU controls the file on the SD card.
9. After control FOTA MCU opens the programming line with Burn_EN pin HIGH.
10. FOTA MCU reads firmware from SD and burns the main MCU.
11. At the end of the burn FOTA controls the process.
12. FOTA MCU sets the Burn_EN pin LOW.
13. FOTA MCU sends the Burn_Done signal to HIGH to un power the FOTA MCU.
14. Update completed.

[image]

* Can u give advice for these scenarios?
* This system works fine?

Thanks for all...

barisdinc, 2020-12-23:
Why this much complicated both in HW and firmware ?
mega2560 can update itself over the cloud easily....

akkoyun, 2020-12-23:
> Why this much complicated both in HW and firmware ? mega2560 can update itself over the cloud easily....
> […](#)

how ?

barisdinc, 2020-12-23:
if your firmware is not more than half of the total memory, you can fit 2
firmware into flash.... Your OTA should overwrite the flash portion which
it is not working....
After flashing itself (by writing directly to flash from cloud) it can
change the next boot to the newly installed portion of the flash...
Then resets itself.. the bootloader checks the next boot firmware position
from eeprom and redirects to that firmware...
You firmware should have a watchdog mechanism to be sure that it is
functioning properly...  if its not working it will rest the device again
then the bootloader should decide to go back
to last working firmware location...

for more you can check multiboot on avr and example bootloaders which
support multiple firmwares on device

best wishes
baris

nickgammon, 2021-03-22:
This "issue" seems out of scope for the sketches I have uploaded in this section. This sounds like a totally new project. I suggest you ask at <http://arduino.cc/forum/> or <http://arduino.stackexchange.com/>.

closed 2021-03-22 by nickgammon