nickgammon/arduino_sketches #78: Reset circuit

Micilera, 2021-11-09:
Hi,
I'm italian, so I'm sorry for my english.
I'm building a board in order to set the fuses of some ATmega328p that I believe I have set in a wrong way and now the chips seem to be died.
My mistake.
I think I can use your project in order to resume the chips. The project does exactly that for what I have understood. That's great.
Looking at the reset circuit I have a concern and I would like to share with you all what is my concern.
There is an output ('VCC' on D5) from the Arduino Uno that goes to the reset circuit on the target board. In the firmware this wire is configured as an output and at initialization time it is set to 'low' (line 271..278 of HV_Parallel_Utils.ino).
Later on, VCC is set to high at line 248, same file.
But if this output is set to low, there is a sort of short circuit between the 'VCC on Target' and GND. 
The short circuit current can be greater than the maximum that the Arduino Uno's output can accept?
Am I making a mistake? May be.
What do you think about?
Thanks for reading.
Best regards,
Guido

[image]

nickgammon, 2021-11-30:
The VCC on the target is *only* connected to D5. It is not also connected to 5V somewhere else. This is necessary because you need to set /RESET to 12V between 20 to 60 µs after powering up the target chip. Thus, D5 powers up the target chip, and the extra circuitry uses a time-delay to bring /RESET to 12V in the appropriate time interval.

nickgammon, 2021-11-30:
For anyone else reading this, the circuit is described [here](http://www.gammon.com.au/forum/?id=12898)

closed 2023-02-21 by nickgammon