nickgammon/arduino_sketches #79: Fixed Filename Uploader - Fusebits

mega-hz, 2021-12-06:
Hello Nick,
firstly thank you for those good uploader-sketches!

I am using the fixed-filename-uploader in a tester/flasher for new 2560 based Boards.
Because of those 2560 are brand new, their Fusebits are set to 1mhz internal as default.
My Problem is: I need to set the Fusebits to run with 16Mhz external crystal.
it seems to be your code reads the fuse-bits and write them back after programming.
How can i set up the fusebits instead of reading them from the chip?
They should be always HI:0xD8, LO:0xFF and EXT:0xFC
So where in the Sketch can i change this?
I can`t find the right variables for the fuses...

Thank you

nickgammon, 2021-12-07:
I had a discussion with someone else about this recently. His solution was as follows:

```c++
if (strcmp (currentSignature.desc, "ATtiny85") == 0)
  {
  writeFuse (0xE2, writeLowFuseByte);
  writeFuse (0xD5, writeHighFuseByte);
  writeFuse (0xFF, writeExtendedFuseByte);
  } 
```

You could do something similar (of course, with a different board name ("ATmega2560"), and appropriate fuse byte values).

I think I would put that code around line 1371 of the sketch, eg. in your case:

```c++
  // now fix up fuses so we can boot    
  if (errors == 0)
    {
    updateFuses (true);

    if (strcmp (currentSignature.desc, "ATmega2560") == 0)
      {
      writeFuse (0xFF, writeLowFuseByte);
      writeFuse (0xD8, writeHighFuseByte);
      writeFuse (0xFC, writeExtendedFuseByte);
      } 

    }
```

mega-hz, 2021-12-07:
Thats it!

thank you very much for your help!
everything runs now greatful

Wolfram.

closed 2021-12-07 by mega-hz

mega-hz, 2021-12-10:
Hi Nick, i have another Problem: If i flash one nee 2560 Board everything is OK. But If i flash the same Board again it doesnt Run anymore. My thoughts are: doesnt be the Chip dont get a chiperase at beginning? If so, where and how can i do this? Does a chiperase clear everything? Also eeprom?
Greetings, Wolfram

nickgammon, 2021-12-10:
>  Does a chiperase clear everything?

Yes.

> Also eeprom?

No.

Since you have changed the fuse to require a 16 MHz clock, do you in fact have a 16 MHz external clock when you try to reflash?

mega-hz, 2021-12-10:
Yes a 16mhz Crystal is on Board

nickgammon, 2021-12-10:
> But If i flash the same Board again it doesnt Run anymore.

You can't program it, or it programs OK (no error messages) but the board doesn't run?

And, to be clear, it runs when you program it the first time?

mega-hz, 2021-12-10:
Programming is OK but it doesnt Run.
At first flashing IT runs after programming

mega-hz, 2021-12-10:
Are you Sure IT makes a chiperase before programming?

nickgammon, 2021-12-10:
See: https://github.com/nickgammon/arduino_sketches/blob/5a5d8966a14878a2c65dc580e47a5941c1264731/Atmega_Hex_Uploader_Fixed_Filename/Atmega_Hex_Uploader_Fixed_Filename.ino#L1032

mega-hz, 2021-12-13:
ok, how can a chip erase include the eeprom?

mega-hz, 2021-12-13:
Hi Nick,

i found out something strange:

 

I have flashed an already flashed 2560, program & verify is ok, but it doesnt run.

After flashing with your program i connected my usb-flasher and started atmel-studio5 ,

loaded the same hexfile and started a verify. I get error message from atmelstudio:

Verifying Flash…Failed! Address=0x0000 expected 0x0c actual 0xFF !

Fusebits and Lockbits are right, LO=FF,HI=D8,EXT=FC and LOCK:FF

 

If i flash it with atmelstudio it runs after flashing!

 

What could be the error? And why does it work fine with a brandnew Controller at firsttime programming?

 

Wolfram

mega-hz, 2021-12-13:
I tried following:

I read the fuses of an unprogrammed controller LO=62, HI=99,EXT=FF

When i write these values to an already programmed chip and then program it with your software,

everything seems fine! Should i change the sketch so those old fuses are programmed bevor the chip-erase comes?

mega-hz, 2021-12-13:
Now i found the reason and the solution:

The HI Fuse must have D9 instead of D8 ! = BOOTRST 
This doesn`t have to be set, because of no bootloader!
Now i can flash every board, even it was flashed before!
- SOLVED -
-