nickgammon/arduino_sketches #80: Changing SD card for external SPI flash memory

AntonioFromBrazil, 2021-12-15:
Hello Nick, on you bootloader from SD card, do you think is possible to change the SD card for an external SPI flash memory such W25Q32 ? Since the SPI flash memory access seam to be easier than a SD card access,   I think that change would be implemented without stress. Is it true ?

nickgammon, 2021-12-18:
Well you can do anything really. However I think the SD card is much easier to manage. You just pop the SD card out, move it over to your computer and put one or more files on it. How would you put the files onto the SPI flash memory?

AntonioFromBrazil, 2021-12-18:
My idea is write the SPI flash memory using a special sketch for this purpouse. Maybe not write SPI flash as files, but modules in binary at fix lenght of 24Kbytes each. After that, the special bootloader would receive a number of module you want to execute... Since you have very cheap SPI ICs as big as 128 Mbit, you would have a wide collection of sketchs stored on it...

nickgammon, 2021-12-18:
Certainly you could change it, however I'll leave that up to you. Basically you need to retain the part that does the programming, remove the stuff which lists the directory and lets you choose a file, and add your own stuff to access the SPI flash memory (and if you are planning to put a wide collection of sketches on it you would need to find the appropriate part).

AntonioFromBrazil, 2021-12-19:
Yes yes... Once the SPI flash was loaded with the various sketches, the idea is to create a special address inside the bootloader that when called move the specific sketch from SPI flash to ATmega flash and execute it. I have only one big problem... I do not know my skill is enough to do such complex job...

nickgammon, 2021-12-21:
OK, but to make that change would basically be a rewrite which goes out of the scope of supporting the existing code. You could try asking on the [Arduino Forum](https://forum.arduino.cc) for help with writing such a sketch.

closed 2021-12-21 by nickgammon