nickgammon/arduino_sketches #82: Atmega_Hex_Uploader filenames

SchneHa, 2022-09-10:
Hello Nick,
if uploading files with Windows 10 to MicroSD card, filenames could be named in different ways: Test1.hex, test2.HEX, TEST3.hex, TEST4.HEX ... Unfortunately it seems not possible, to rename filename extensions like .hex to upper case extensions in Windows 10.
The sketch only displays filenames with .HEX (uppercase) no other. Thus, if you know the filenames of the other files they can be choosen and flashed.
It would be easier to use if you convert ALL filenames to upper case filenames and handle it. Maybe it isn't so much effort to change the sketch.
Best regards,
Hans

harryberlin, 2022-09-10:
in Windows you have to rename in steps before you would swap to upper case.
XYZ.hex -> XYZ.he -> XYZ.HEX

SchneHa, 2022-09-10:
Thank you for this hint but it would be easier if the sketch would recognize all filenames no matter if they're lower or upper case.

nickgammon, 2023-02-21:
You can change the sketch yourself to do this.

closed 2023-02-21 by nickgammon