nickgammon/arduino_sketches #83: Atmega_Hex_Uploader_Fixed_Filename: How to change the fuse setting?

kapil-iitk, 2022-11-23:
Dear Nick Gammon,

Thank you, nice uploader. Can you please share how to change the fuse setting for a particular microcontroller through your program? The programmer is working fine, and I can upload the file firmware.hex into the microcontroller. However, for the new microcontroller, if I am uploading the program for 12 MHz, the microcontroller is not working at 12 MHz (External). Probably it's because of the factory set at 1 MHz (Internal). Therefore can you please point me to which portion is responsible for writing the fuse?

(I am using Atmega_Hex_Uploader_Fixed_Filename.ino due to its simplicity and no-computer requirement)

Thank you in advance.

nickgammon, 2022-11-25:
I'm not sure I understand the question. There is no fuse that specifies what speed the processor is running at. 

However there is a function in the code `writeFuse` which can be used to program a fuse.

kapil-iitk, 2022-11-25:
Thank you for your reply, and sorry for the confusion. I am using a microcontroller having factory installed fuse. And I wish to program the microcontroller along with wanted to write the fuse. I was thinking about this function (writeFuse) as it was there. However, I am not sure how to use it and where to use it to change the fuse setting. Can you please share this? I would appreciate it if you could share or indicate which line and where they have to be placed if I have to write fuse bit to the microcontroller.

Thank you.

Regards

nickgammon, 2022-11-27:
Which fuse are you planning to change, and what value are you planning to write to it?

`updateFuses` calculates where the bootloader starts and sets the correct fuse to the correct value so that it boots correctly.

`updateFuses` calls `writeFuse`.

kapil-iitk, 2022-11-27:
Dear,

Thank you for your reply regarding updateFuses. For the new atmega8 microcontroller fuse setting is HFuse 0xD9 LFuse 0xE1 (This we have operating at internal 1 MHz). After uploading the program to the microcontroller, I wanted to change the default fuse setting to HFuse 0xD9 and LFuse 0xEF (so I could use an external crystal 12 MHz). 

Based on my understanding, in code (https://github.com/nickgammon/arduino_sketches/blob/master/Atmega_Hex_Uploader_Fixed_Filename/Atmega_Hex_Uploader_Fixed_Filename.ino) after "bool ok = writeFlashContents ();" (Line no. 1470), I have to include the following lines,

writeFuse (fuses [highFuse], 0xD9);
writeFuse (fuses [lowFuse], 0xEF);

Is this correct?

nickgammon, 2022-11-27:
No. If you look at the declaration for writeFuse you will see:

```
void writeFuse (const byte newValue, const byte instruction)
```

Clearly the new value (the thing you want to write) is the first argument, not the second one.

And, the instructions are declared here:

```
// which program instruction writes which fuse
const byte fuseCommands [4] = { writeLowFuseByte, writeHighFuseByte, writeExtendedFuseByte, writeLockByte };
```

So your code would look like this:

```
writeFuse (0xD9, writeHighFuseByte);
writeFuse (0xEF, writeLowFuseByte);
```

I also would not do it where you suggested, that function might return if you had not selected a file, and then you are still writing the fuses. I suggest at line 1370, change:

```
  // now fix up fuses so we can boot    
  if (errors == 0)
    updateFuses (true);
```

to:

```
  // now fix up fuses so we can boot    
  if (errors == 0)
    {
    // updateFuses (true);  // OMITTED BECAUSE OF TWO LINES BELOW
    writeFuse (0xD9, writeHighFuseByte);
    writeFuse (0xEF, writeLowFuseByte);
    }
```

kapil-iitk, 2022-11-28:
Thank you very much for your response and help. I will test it with some modifications.

kapil-iitk, 2022-11-29:
Dear Nick,

Thank you for your support. While exploring your code, I thought that by slight modification, your program could do it for all kinds of (AVR!) microcontrollers (writing fuse bit). I modified the code and included two (generic functions). Can you please comment on the flowing 

 // Function for string (from the text file placed inside SD card) to hex
byte StrToHex(char str[])
{
  return (byte) (str <= '9') ? str - '0' : str - '7';
}

// Function to check if file exist
boolean fileExists(const char* name) {
  return sd.exists(name);
}

//Line number 1370
 if (errors == 0)
    {
      if(fileExists(fileName_fuse)){

        // Modified for kapil
        // fileName_fuse is a file named "fuse.txt"
        // contain single line "DEFF0500"
        // HFuseLFuseEFuesLock
       
        File NewFile;
        String read_line;
        char hexValue[10];
        byte Fuse_Byte_file[4];
        int pp=0;
   
        NewFile = sd.open(fileName_fuse);
        if (NewFile){
          while(NewFile.available()){
          read_line = NewFile.readStringUntil('\n');
          break;
          }
        }
        NewFile.close();      
        read_line.toCharArray(hexValue,10);
       
        for(int f_line=0; f_line<4; f_line++){
          Fuse_Byte_file[f_line] = (16 * StrToHex(hexValue[pp])) + StrToHex(hexValue[pp+1]);
          //Serial.println(Fuse_Byte_file[f_line], HEX);
          //Serial.println("\n"); // it displays perfectly
          pp=pp+2;
        }      
        // For USBASP following fusebit are used for atmega8    HFUSE=0xc9  LFUSE=0xef
        //ATmega8 HFUSE=0xD9  LFUSE=0xE1  Factory Default Settings for ATmega8.
        //H fuse: Reset Disabled  : Unprogrammed (Reset pin is enabled)
        //Watchdog        : Off
        //SPI Programming : Enabled
        //CKOPT           : Unprogrammed
        //EEPROM Preserve : Unprogrammed (EEPROM not preserved)
        //BOOT size       : 1024 words
        //Reset Vector    : User flash code
        //L fuse: Brown-out level : 2.7V
        //Brown-out detect: Disabled
        //Startup timing  : Default
        //Clock Source    : Internal, 1MHz
        //ATmega8 HFUSE=0xD9  LFUSE=0xEF ; External Crystal upto 16MHz. Rest of the options : Default.
        writeFuse (Fuse_Byte_file[0], writeHighFuseByte); // For arduino Uno HFUSE: 0XDE
        writeFuse (Fuse_Byte_file[1], writeLowFuseByte); // For arduino Uno LFUSE: 0xFF
        writeFuse (Fuse_Byte_file[2], writeExtendedFuseByte); // For arduino Uno EFUSE: 0x05
        //writeFuse (Fuse_Byte_file[3], writeLockByte); //Caution!! use carefully
      }
      else
        updateFuses (true);
    }

Can you please comment on this modification (else I have to make AVR fuse doctor as well :-) ) before I actually implement it.

nickgammon, 2023-02-21:
Looks OK.

closed 2023-02-21 by nickgammon