nickgammon/arduino_sketches #84: Cannot enter programming mode in target chip

ovehbe, 2023-02-11:
Hi, I've been using "Atmega_Hex_Uploader_Fixed_Filename" version for while which was working great and thanks a lot for that! But after some time when I tried to use it again I got the 3xRed error. Which is very wierd and I've been trying to figure out whats happening for almost 4 hours now. I've checked the wirings a million times tried different power sources checked cables for damage with a multimeter tried a new sd card module tried a new sd card and a new hex file I tried with 2 different arduinos but nope not entering programming mode 😅. Also checked every issue at github. Buy any chance someone might help me with this? the chip on the arduino is ATMEGA328P.

nickgammon, 2023-02-12:
If it was working, but isn't now, it is likely you have changed the fuses on the chip you are trying to program. One possible change is to make it expect an external crystal, which means it won't work unless you supply a clock signal to the clock pin. Or, you may have disabled the ability to reset the chip.

Try connecting the chip to another spare Arduino and load up the https://github.com/nickgammon/arduino_sketches/tree/master/Atmega_Board_Detector sketch.

You need six wires between the Arduino and the chip in question. That will tell you what the fuses are on the target chip.

If that doesn't work, try a different chip, to prove that your wiring is OK.

ovehbe, 2023-02-13:
Hi, first of all thanks for the response. Second, I ran the board detector sketch and here is the output:

```
Atmega chip detector.
Written by Nick Gammon.
Version 1.20
Compiled on Feb 13 2023 at 04:26:44 with Arduino IDE 10819.
Attempting to enter ICSP programming mode ...
Entered programming mode OK.
Signature = 0x1E 0x95 0x0F 
Processor = ATmega328P
Flash memory size = 32768 bytes.
LFuse = 0xFF 
HFuse = 0xDE 
EFuse = 0xFD 
Lock byte = 0xCF 
Clock calibration = 0xC2 
Bootloader in use: Yes
EEPROM preserved through erase: No
Watchdog timer always on: No
Bootloader is 512 bytes starting at 7E00

Bootloader:

7E00: 0x11 0x24 0x84 0xB7 0x14 0xBE 0x81 0xFF 0xF0 0xD0 0x85 0xE0 0x80 0x93 0x81 0x00 
7E10: 0x82 0xE0 0x80 0x93 0xC0 0x00 0x88 0xE1 0x80 0x93 0xC1 0x00 0x86 0xE0 0x80 0x93 
7E20: 0xC2 0x00 0x80 0xE1 0x80 0x93 0xC4 0x00 0x8E 0xE0 0xC9 0xD0 0x25 0x9A 0x86 0xE0 
7E30: 0x20 0xE3 0x3C 0xEF 0x91 0xE0 0x30 0x93 0x85 0x00 0x20 0x93 0x84 0x00 0x96 0xBB 
7E40: 0xB0 0x9B 0xFE 0xCF 0x1D 0x9A 0xA8 0x95 0x81 0x50 0xA9 0xF7 0xCC 0x24 0xDD 0x24 
7E50: 0x88 0x24 0x83 0x94 0xB5 0xE0 0xAB 0x2E 0xA1 0xE1 0x9A 0x2E 0xF3 0xE0 0xBF 0x2E 
7E60: 0xA2 0xD0 0x81 0x34 0x61 0xF4 0x9F 0xD0 0x08 0x2F 0xAF 0xD0 0x02 0x38 0x11 0xF0 
7E70: 0x01 0x38 0x11 0xF4 0x84 0xE0 0x01 0xC0 0x83 0xE0 0x8D 0xD0 0x89 0xC0 0x82 0x34 
7E80: 0x11 0xF4 0x84 0xE1 0x03 0xC0 0x85 0x34 0x19 0xF4 0x85 0xE0 0xA6 0xD0 0x80 0xC0 
7E90: 0x85 0x35 0x79 0xF4 0x88 0xD0 0xE8 0x2E 0xFF 0x24 0x85 0xD0 0x08 0x2F 0x10 0xE0 
7EA0: 0x10 0x2F 0x00 0x27 0x0E 0x29 0x1F 0x29 0x00 0x0F 0x11 0x1F 0x8E 0xD0 0x68 0x01 
7EB0: 0x6F 0xC0 0x86 0x35 0x21 0xF4 0x84 0xE0 0x90 0xD0 0x80 0xE0 0xDE 0xCF 0x84 0x36 
7EC0: 0x09 0xF0 0x40 0xC0 0x70 0xD0 0x6F 0xD0 0x08 0x2F 0x6D 0xD0 0x80 0xE0 0xC8 0x16 
7ED0: 0x80 0xE7 0xD8 0x06 0x18 0xF4 0xF6 0x01 0xB7 0xBE 0xE8 0x95 0xC0 0xE0 0xD1 0xE0 
7EE0: 0x62 0xD0 0x89 0x93 0x0C 0x17 0xE1 0xF7 0xF0 0xE0 0xCF 0x16 0xF0 0xE7 0xDF 0x06 
7EF0: 0x18 0xF0 0xF6 0x01 0xB7 0xBE 0xE8 0x95 0x68 0xD0 0x07 0xB6 0x00 0xFC 0xFD 0xCF 
7F00: 0xA6 0x01 0xA0 0xE0 0xB1 0xE0 0x2C 0x91 0x30 0xE0 0x11 0x96 0x8C 0x91 0x11 0x97 
7F10: 0x90 0xE0 0x98 0x2F 0x88 0x27 0x82 0x2B 0x93 0x2B 0x12 0x96 0xFA 0x01 0x0C 0x01 
7F20: 0x87 0xBE 0xE8 0x95 0x11 0x24 0x4E 0x5F 0x5F 0x4F 0xF1 0xE0 0xA0 0x38 0xBF 0x07 
7F30: 0x51 0xF7 0xF6 0x01 0xA7 0xBE 0xE8 0x95 0x07 0xB6 0x00 0xFC 0xFD 0xCF 0x97 0xBE 
7F40: 0xE8 0x95 0x26 0xC0 0x84 0x37 0xB1 0xF4 0x2E 0xD0 0x2D 0xD0 0xF8 0x2E 0x2B 0xD0 
7F50: 0x3C 0xD0 0xF6 0x01 0xEF 0x2C 0x8F 0x01 0x0F 0x5F 0x1F 0x4F 0x84 0x91 0x1B 0xD0 
7F60: 0xEA 0x94 0xF8 0x01 0xC1 0xF7 0x08 0x94 0xC1 0x1C 0xD1 0x1C 0xFA 0x94 0xCF 0x0C 
7F70: 0xD1 0x1C 0x0E 0xC0 0x85 0x37 0x39 0xF4 0x28 0xD0 0x8E 0xE1 0x0C 0xD0 0x85 0xE9 
7F80: 0x0A 0xD0 0x8F 0xE0 0x7A 0xCF 0x81 0x35 0x11 0xF4 0x88 0xE0 0x18 0xD0 0x1D 0xD0 
7F90: 0x80 0xE1 0x01 0xD0 0x65 0xCF 0x98 0x2F 0x80 0x91 0xC0 0x00 0x85 0xFF 0xFC 0xCF 
7FA0: 0x90 0x93 0xC6 0x00 0x08 0x95 0x80 0x91 0xC0 0x00 0x87 0xFF 0xFC 0xCF 0x80 0x91 
7FB0: 0xC0 0x00 0x84 0xFD 0x01 0xC0 0xA8 0x95 0x80 0x91 0xC6 0x00 0x08 0x95 0xE0 0xE6 
7FC0: 0xF0 0xE0 0x98 0xE1 0x90 0x83 0x80 0x83 0x08 0x95 0xED 0xDF 0x80 0x32 0x19 0xF0 
7FD0: 0x88 0xE0 0xF5 0xDF 0xFF 0xCF 0x84 0xE1 0xDE 0xCF 0x1F 0x93 0x18 0x2F 0xE3 0xDF 
7FE0: 0x11 0x50 0xE9 0xF7 0xF2 0xDF 0x1F 0x91 0x08 0x95 0x80 0xE0 0xE8 0xDF 0xEE 0x27 
7FF0: 0xFF 0x27 0x09 0x94 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0x04 0x04 

MD5 sum of bootloader = 0xFB 0xF4 0x9B 0x7B 0x59 0x73 0x7F 0x65 0xE8 0xD0 0xF8 0xA5 0x08 0x12 0xE7 0x9F 
Bootloader name: optiboot_atmega328

First 256 bytes of program memory:

00: 0x0C 0x94 0x61 0x00 0x0C 0x94 0x7E 0x00 0x0C 0x94 0x7E 0x00 0x0C 0x94 0x7E 0x00 
10: 0x0C 0x94 0x7E 0x00 0x0C 0x94 0x7E 0x00 0x0C 0x94 0x7E 0x00 0x0C 0x94 0x7E 0x00 
20: 0x0C 0x94 0x7E 0x00 0x0C 0x94 0x7E 0x00 0x0C 0x94 0x7E 0x00 0x0C 0x94 0x7E 0x00 
30: 0x0C 0x94 0x7E 0x00 0x0C 0x94 0x7E 0x00 0x0C 0x94 0x7E 0x00 0x0C 0x94 0x7E 0x00 
40: 0x0C 0x94 0x9D 0x00 0x0C 0x94 0x7E 0x00 0x0C 0x94 0x7E 0x00 0x0C 0x94 0x7E 0x00 
50: 0x0C 0x94 0x7E 0x00 0x0C 0x94 0x7E 0x00 0x0C 0x94 0x7E 0x00 0x0C 0x94 0x7E 0x00 
60: 0x0C 0x94 0x7E 0x00 0x0C 0x94 0x7E 0x00 0x00 0x00 0x00 0x00 0x24 0x00 0x27 0x00 
70: 0x2A 0x00 0x00 0x00 0x00 0x00 0x25 0x00 0x28 0x00 0x2B 0x00 0x00 0x00 0x00 0x00 
80: 0x23 0x00 0x26 0x00 0x29 0x00 0x04 0x04 0x04 0x04 0x04 0x04 0x04 0x04 0x02 0x02 
90: 0x02 0x02 0x02 0x02 0x03 0x03 0x03 0x03 0x03 0x03 0x01 0x02 0x04 0x08 0x10 0x20 
A0: 0x40 0x80 0x01 0x02 0x04 0x08 0x10 0x20 0x01 0x02 0x04 0x08 0x10 0x20 0x00 0x00 
B0: 0x00 0x07 0x00 0x02 0x01 0x00 0x00 0x03 0x04 0x06 0x00 0x00 0x00 0x00 0x00 0x00 
C0: 0x00 0x00 0x11 0x24 0x1F 0xBE 0xCF 0xEF 0xD8 0xE0 0xDE 0xBF 0xCD 0xBF 0x11 0xE0 
D0: 0xA0 0xE0 0xB1 0xE0 0xEA 0xE5 0xF4 0xE0 0x02 0xC0 0x05 0x90 0x0D 0x92 0xA2 0x30 
E0: 0xB1 0x07 0xD9 0xF7 0x11 0xE0 0xA2 0xE0 0xB1 0xE0 0x01 0xC0 0x1D 0x92 0xAB 0x30 
F0: 0xB1 0x07 0xE1 0xF7 0x0E 0x94 0x1B 0x02 0x0C 0x94 0x2B 0x02 0x0C 0x94 0x00 0x00 

Programming mode off.
```
Also I didn't really understand the firs paragraph of your response much to be honest. But I know this I program lots of arduinos and I noticed the arduinos change sometimes I use clones but not every batch is the same.

nickgammon, 2023-02-13:
Did you provide a clock pulse to the chip? It is configured to require a crystal, see <https://www.engbedded.com/fusecalc/>.

ovehbe, 2023-02-13:
No, by any chance you can explain a bit more or point me into some explanation?

nickgammon, 2023-02-13:
Your fuses require a crystal to the target chip. Have you installed one? Also, do you have decoupling capacitors on the +5 power rails?

ovehbe, 2023-02-13:
No I don't have neither, Still not sure about the fuse. But should I try adding the capacitators? If you have any wiring diagram that would be great.

nickgammon, 2023-02-14:
Any attempt to run microprocessors should have decoupling capacitors on the supply rails to smooth out spikes caused by program execution. These should be as close as possible to Vcc pin (and AVcc on the Atmega328P) as shown in this image:

[image]

nickgammon, 2023-02-14:
As mentioned [here](http://www.gammon.com.au/forum/?id=11638):

>  If the target device requires an external clock the sketch provides an 8 MHz clock output on pin D9, so connect pin D9 to the clock input (pin 9 on an Atmega328).

It is also mentioned in the comments inside the sketch:

```c++

/*
Connect target processor like this:
  D4: (SCK)   --> SCK as per datasheet
  D5: (SS)    --> goes to /RESET on target
  D6: (MISO)  --> MISO as per datasheet
  D7: (MOSI)  --> MOSI as per datasheet
  
  D9: 8 Mhz clock signal if required by target

*/
```

In other words, D9 (Arduino board digital pin 9) on the board running the "Atmega_Hex_Uploader_Fixed_Filename.ino" sketch is c connected to processor pin 9 on the Atmega328P. This provides an 8 MHz clock signal to it.

closed 2023-02-21 by nickgammon