nickgammon/arduino_sketches #85: Atmega_Hex_Uploader need old SdFat version to compile

noeldum, 2023-05-19:
First of all thank you Nick for the brilliant work done with this.  I have tried various solutions for a stand alone programmer and this is the most serious work done on this here.

I tried to compile the Atmega_Hex_Uploader sketch and it came with errors. I was using SdFat version 2.2.2.

I downgraded the SdFAt  library to version 1.1.4.
https://github.com/greiman/SdFat/releases/tag/1.1.4

A quick fix for this might be to indicate the version required next to the library link.

A nicer solution would be for someone to adapt the sketch for the latest version of the library.

nickgammon, 2023-05-20:
In [my post about the uploader](http://www.gammon.com.au/forum/?id=11638) I mention that it doesn't work with the newer SdFat library, and give a link to the one it works with.

noeldum, 2023-05-20:
Thank you. I missed the link on the page.
It's probably good to know a link on the original repo release too I guess. That's the latest I could find to work above.

I am trying to flash both bootloader and sketch at the same time. Would you be able to help in the thread below?
https://forum.arduino.cc/t/perserving-bootloader-while-programming-from-spi/1128912/3

closed 2023-05-24 by noeldum