nickgammon/arduino_sketches #88: Arduino 1.18 and Arduino 2.x give ino file const error! Does not work!

Alectronics, 2023-08-20:
\Atmega_Board_Programmer\Atmega_Board_Programmer\bootloader_atmega328.h:3:34: error: variable 'atmega328_optiboot' must be const in order to be put into read-only section by means of '__attribute__((progmem))'
 byte PROGMEM atmega328_optiboot [] = {
                                  ^
C:\Users\Albert van Bemmelen\Desktop\Nano3.0 modules dispute!\Atmega_Board_Programmer\Atmega_Board_Programmer\Atmega_Board_Programmer.ino:147:9: error: 'atmega168_optiboot' was not declared in this scope
         atmega168_optiboot,   // loader image
         ^~~~~~~~~~~~~~~~~~
C:\Users\Albert van Bemmelen\Desktop\Nano3.0 modules dispute!\Atmega_Board_Programmer\Atmega_Board_Programmer\Atmega_Board_Programmer.ino:147:9: note: suggested alternative: 'atmega328_optiboot'
         atmega168_optiboot,   // loader image
         ^~~~~~~~~~~~~~~~~~
         atmega328_optiboot
C:\Users\Albert van Bemmelen\Desktop\Nano3.0 modules dispute!\Atmega_Board_Programmer\Atmega_Board_Programmer\Atmega_Board_Programmer.ino:149:16: error: 'atmega168_optiboot' was not declared in this scope
         sizeof atmega168_optiboot,
                ^~~~~~~~~~~~~~~~~~
C:\Users\Albert van Bemmelen\Desktop\Nano3.0 modules dispute!\Atmega_Board_Programmer\Atmega_Board_Programmer\Atmega_Board_Programmer.ino:149:16: note: suggested alternative: 'atmega328_optiboot'
         sizeof atmega168_optiboot,
                ^~~~~~~~~~~~~~~~~~
                atmega328_optiboot
C:\Users\Albert van Bemmelen\Desktop\Nano3.0 modules dispute!\Atmega_Board_Programmer\Atmega_Board_Programmer\Atmega_Board_Programmer.ino:184:9: error: 'optiboot_atmega1280_hex' was not declared in this scope
         optiboot_atmega1280_hex,
         ^~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Albert van Bemmelen\Desktop\Nano3.0 modules dispute!\Atmega_Board_Programmer\Atmega_Board_Programmer\Atmega_Board_Programmer.ino:186:16: error: 'optiboot_atmega1280_hex' was not declared in this scope
         sizeof optiboot_atmega1280_hex,
                ^~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Albert van Bemmelen\Desktop\Nano3.0 modules dispute!\Atmega_Board_Programmer\Atmega_Board_Programmer\Atmega_Board_Programmer.ino:195:9: error: 'atmega2560_bootloader_hex' was not declared in this scope
         atmega2560_bootloader_hex,// loader image
         ^~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Albert van Bemmelen\Desktop\Nano3.0 modules dispute!\Atmega_Board_Programmer\Atmega_Board_Programmer\Atmega_Board_Programmer.ino:197:16: error: 'atmega2560_bootloader_hex' was not declared in this scope
         sizeof atmega2560_bootloader_hex,
                ^~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Albert van Bemmelen\Desktop\Nano3.0 modules dispute!\Atmega_Board_Programmer\Atmega_Board_Programmer\Atmega_Board_Programmer.ino:217:9: error: 'optiboot_atmega1284p_hex' was not declared in this scope
         optiboot_atmega1284p_hex,
         ^~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Albert van Bemmelen\Desktop\Nano3.0 modules dispute!\Atmega_Board_Programmer\Atmega_Board_Programmer\Atmega_Board_Programmer.ino:219:16: error: 'optiboot_atmega1284p_hex' was not declared in this scope
         sizeof optiboot_atmega1284p_hex,
                ^~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Albert van Bemmelen\Desktop\Nano3.0 modules dispute!\Atmega_Board_Programmer\Atmega_Board_Programmer\Atmega_Board_Programmer.ino:235:9: error: 'atmega8_hex' was not declared in this scope
         atmega8_hex,
         ^~~~~~~~~~~
C:\Users\Albert van Bemmelen\Desktop\Nano3.0 modules dispute!\Atmega_Board_Programmer\Atmega_Board_Programmer\Atmega_Board_Programmer.ino:237:16: error: 'atmega8_hex' was not declared in this scope
         sizeof atmega8_hex,
                ^~~~~~~~~~~
C:\Users\Albert van Bemmelen\Desktop\Nano3.0 modules dispute!\Atmega_Board_Programmer\Atmega_Board_Programmer\Atmega_Board_Programmer.ino: In function 'void writeBootloader()':
C:\Users\Albert van Bemmelen\Desktop\Nano3.0 modules dispute!\Atmega_Board_Programmer\Atmega_Board_Programmer\Atmega_Board_Programmer.ino:419:20: error: 'ATmegaBOOT_168_atmega328_pro_8MHz_hex' was not declared in this scope
       bootloader = ATmegaBOOT_168_atmega328_pro_8MHz_hex;
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

exit status 1

Compilation error: variable 'atmega328_optiboot' must be const in order to be put into read-only section by means of '__attribute__((progmem))'

nickgammon, 2023-08-20:
I can't reproduce that. With both Arduino 1.18 and Arduino 2.1.1 it compiles OK:

```
Sketch uses 27196 bytes (84%) of program storage space. Maximum is 32256 bytes.
Global variables use 242 bytes (11%) of dynamic memory, leaving 1806 bytes for local variables. Maximum is 2048 bytes.
```

The error message you quoted says that the variable had to be **const** which it is:

https://github.com/nickgammon/arduino_sketches/blob/master/Atmega_Board_Programmer/bootloader_atmega328.h

```
const byte atmega328_optiboot [] PROGMEM = {
```

Have you downloaded the latest version of the code from GitHub?

Alectronics, 2023-08-21:
My version didn't have const in it.
So probably I had an older one? But because after I added a const het still didn't work.

I will check it after I'm back from work. I had 4 not working Nano 3.0 from China why it cost me my weekend trying the only blinking dead Nano CH340C/G modules getting to work. Why your program was very interesting.

Best regards,
Albert.

https://github.com/nickgammon/arduino_sketches/blob/master/Atmega_Board_Programmer/bootloader_atmega328.h

nickgammon, 2023-08-21:
There are a number of such files. Since you are posting a bug report here on Github, I assumed you would have the latest code. I suggest downloading all the code again (from here).

Alectronics, 2023-08-21:
Hi Nick.

There is something strange about your packed sketch file. When I extract these with WinRar they never show up anywhere in Windows 10? Looked in older 1.8.12 and also in Arduino 2.1.

Why I probably only used a bad part of those files previously?

Best regards.
Albert.

Alectronics, 2023-08-21:
Apparently Arduino still wasn't able to unlock my bad Arduino Nano 3.0 modules over ISP with my Pololu v2.1 programmer as STK500 ISP working. I loaded the working Pololu configuration program see attached screencopy. And I see clearly that my pololu over ISP is trying to flash my dead (only flashing L led and lit 5V PWR led) Nano boards from China.

The -F error keeps messing with the flashing process. Please see also that attached screencopy.
I also upgraded to the newest Arduino 2.1 today but didn't help. Neither with the old or new 328 bootloader selection.

Albert.

Alectronics, 2023-08-21:
I solved the issue but couldn't use your program sadly to solve the issue I had.

My boards were dead not communicating at all!
But after 2 days of dismay I found how to solve the 4 useless Nano 3.0 boards issue.

luckily I found and solved the problem today. With my TL866II I flashed the new optiboot Arduino 328p bootfile from ARDUINO IDE over into the bad not communicating only flashing L led and lit PWR led NANO 3.0 defect looking boards. ISP connected. That previously were not communicating over USB and neither on ISP!
BUT WHILE AND DURING THE PROGRAMMING the button on the NANO 3.0 module must be pressed in all the time! Else the Nano 3.0 boards are completely dead and are wrong read as being some strange $FFFFFFFF processor! Now the modules are working and correctly read as 0x1E950F processor boards! And can without any problem be used in any ARDUINO IDE program.

Best regards.
Albert.

nickgammon, 2023-08-21:
> There is something strange about your packed sketch file. When I extract these with WinRar they never show up anywhere in Windows 10? Looked in older 1.8.12 and also in Arduino 2.1.

I don't know why you are using WinRar. They aren't RAR files. They are ZIP files which Windows 10 should easily be able to open natively. It is a very common format. See screenshot:

[image]

Alectronics, 2023-08-22:
Thanks Nick but I solved my 4 not working Nano 3.0 boards yesterday.

luckily I found and solved the problem. With my TL866II I flashed the new optiboot Arduino 328p bootfile from ARDUINO IDE over into the bad not communicating only flashing L led and lit PWR led NANO 3.0 defect looking boards. That were not communicating over USB and neither on ISP!
BUT WHILE AND DURING THE PROGRAMMING the button on the NANO 3.0 module must be pressed in all the time! Else the Nano 3.0 boards are completely dead and are wrong read as being some strange $FFFFFFFF processor! Now the modules are working and correctly read as Vid: 0x1A86 processor and Pid:0x7523 boards! And can without any problem be used in any ARDUINO IDE program.

Best regards.

Albert.

[Screenshot from 2023-08-22 07-52-08]<https://user-images.githubusercontent.com/185088/262162609-86674926-29d3-4a8a-936c-76bf7e033379.png>