nickgammon/arduino_sketches #89: Incorrect Fuse manipulation of BOOTRST

Kingbuzz0, 2024-05-24:
Hi,

I have a binary for the Atmega328 family (in this example, ATmega328PB) that has no bootloader.  As such, I explicitly set the HFuse Bit 0 (BOOTRST) to b'0.   I noticed that even though your Atmega_Hex_Uploader_Fixed_Filename sketch correctly detects that hex file starts at address 0 and contains no bootloader it forcibly changes BOOTRST to b'1.   It seems that this would cause an error on reset if there is no bootloader, but the reset vector points to whatever is set for BOOTSZ0/1 bits.  

The code in question starts at line 1214:

  if (lowestAddress == 0)
    {
    // don't use bootloader  
    fuses [fusenumber] |= 1;
    }

I can also confirm the behavior is the same for the Atmega_Hex_Uploader sketch.

Thanks,
Rob

nickgammon, 2024-05-25:
Have you tested this and found it does not work? The fuses work backwards to what you might expect. See the data sheet below.

[image]

Writing 1 to BOOTRST means it will start executing at address 0, which is what you want.

Kingbuzz0, 2024-05-25:
Well hot damn.  You are absolutely correct and I have had this backwards for quite a long time.   In fact I had not tested the correct output, but instead I have a handful of devices running the backward way for years perfectly fine.   

Before this is closed out, do you have an guess how incorrectly pointing the reset vector a non-existent boot loader address could still have perfectly functioning code?

nickgammon, 2024-05-25:
The bootloader area probably had hex FF in it, I'm not sure what that does in machine code, but assuming it does not send the processor into a loop it would have eventually gone to the end of memory and then reached address 0 and started executing your code

nickgammon, 2024-05-25:
The fuses are programmed by "blowing" them, which means the programmed state is 0, and non-programmed is 1.

Kingbuzz0, 2024-05-25:
Thanks for the thoughts.

closed 2024-05-25 by Kingbuzz0